labapart/gattlib #6: CMake error on raspberry pi 3

dwaq, 2017-01-29:
I can make it to the CMake step, but then I receive this error:

```
-- Checking for one of the modules 'bluez'
CMake Error at /usr/share/cmake-3.6/Modules/FindPkgConfig.cmake:646 (message):
  None of the required 'bluez' found
Call Stack (most recent call first):
  CMakeLists.txt:56 (pkg_search_module)

CMake Error at CMakeLists.txt:62 (list):
  list GET given empty list

CMake Error at CMakeLists.txt:63 (list):
  list GET given empty list

-- Configuring incomplete, errors occurred!
```
My bluez version is 5.23. I can't tell from your README if this is unsupported or I need to be on 4.1xx.
```
dpkg --status bluez | grep '^Version:'
Version: 5.23-2+rpi2
```

Thanks!

oliviermartin, 2017-01-30:
The issue looks to be the missing package `libbluetooth-dev` on your system. CMake complains about `None of the required 'bluez' found`.
I would have thought CMake would have stopped at this line.

Try again after installing `libbluetooth-dev`: `sudo apt-get install libbluetooth-dev`

bjornspockeli, 2017-01-30:
Hi dwaq, I had the same issue. Looking at the latest release( v0.2) [changelog ](https://github.com/labapart/gattlib/releases)it would seem that gattlib only supports bluez v4.101 and v4.99.

oliviermartin, 2017-01-30:
It might work but I have not tested. I would be more than happy gattlib to be bluez version independent.
I think early version of bluez 5.x might still work but I have not tested it.

oliviermartin, 2017-01-30:
I updated the documentation (https://github.com/labapart/gattlib/commit/7f32f5c346f2247075cf41950ed14e1309946c39), you might also need `libreadline-dev`.  
I might do more testing with Bluez 5.xx in the next few days.

bjornspockeli, 2017-01-30:
I just tried running the build/examples/ble_scan example from my Raspberry Pi running Jessie and Bluez v5.23 with a BLE device advertising nearby and the example finished immediatly.

```
pi@raspberrypi:~/code/ble/include/gattlib/build/examples/ble_scan $ sudo ./ble_scan
Scan completed
```
If I use `sudo hcitool lescan` multiple devices are listed, so this indicates that gattlib and Bluez v5.23 isnt compatible.

oliviermartin, 2017-01-31:
I have just added support for Bluez 5 (based on Bluez 5.43). I have not fully tested so I will be happy to get feedback! I would particularly be happy to know if it works on Bluez 5.23.

bjornspockeli, 2017-01-31:
I pulled the repo and ran make, but I got the following error.

```
[ 80%] Linking C executable ble_scan
../../libgattlib.so: undefined reference to `bswap_128'
collect2: error: ld returned 1 exit status
examples/ble_scan/CMakeFiles/ble_scan.dir/build.make:95: recipe for target 'examples/ble_scan/ble_scan' failed
make[2]: *** [examples/ble_scan/ble_scan] Error 1
CMakeFiles/Makefile2:122: recipe for target 'examples/ble_scan/CMakeFiles/ble_scan.dir/all' failed
make[1]: *** [examples/ble_scan/CMakeFiles/ble_scan.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

oliviermartin, 2017-01-31:
Can you run these commands: `rm -Rf * && cmake .. & make VERBOSE=1` and send me the logs?

bjornspockeli, 2017-01-31:
Done, here is the log. Sorry for the formatting.
[gattlib_make_log.txt](https://github.com/labapart/gattlib/files/743373/gattlib_make_log.txt)

oliviermartin, 2017-01-31:
I am not sure why I does not build on your computer. `bswap_128` is defined as `static inline` in `lib/bluetooth.h` and all the files calling this function include this header.

dwaq, 2017-02-01:
After installing `libbluetooth-dev`, I'm having the same issue as @bjornspockeli with the "`undefined reference to `bswap_128'`". My log output is the same as well. I'll try to dig in a little deeper to see what the issue could be.

oliviermartin, 2017-02-01:
Interestingly both of you are trying to build from the Raspberry Pi while I am building from my x86 laptop.
One way to debug it as well would be to rename every single `bswap_128` into `bswap_128_1`, `bswap_128_2`, `bswap_128_3`, etc and also create the respective inline functions. We could see which `bswap_128` raises the issue. I am guessing all of them.

bjornspockeli, 2017-02-01:
I do not know if this is relevant or related for that matter, but I tried compile gattlib with `make -j 4` so that the build went a bit faster since im doing this on the RPi. This caused the compiler to complain about the undefined reference to bswap_128, but it also complained about some for loops

`‘for’ loop initial declarations are only allowed in C99 or C11 mode`

see the attached make log.
[make_j4_verbose_log.txt](https://github.com/labapart/gattlib/files/746192/make_j4_verbose_log.txt)

oliviermartin, 2017-02-08:
Thanks @bjornspockeli, I only saw your update today as Github does not seem to send email when users update their message. I hopefully fixed the C90 issues (I guess GCC on Rasberry Pi 3 builds for C90).
I have not used your workaround for `bswap_128` as it only happens on Rasberry Pi. I am a bit worry if I use it then it will raise an error of duplicate definitions on the other platforms.

Can you confirm `bluetooth.h` contains `bswap_128` definition? On my x86 latop:

```
$ find /usr/include -name 'bluetooth.h' -exec grep -H bswap_128 {} \;
/usr/include/bluetooth/bluetooth.h:static inline void bswap_128(const void *src, void *dst)
/usr/include/bluetooth/bluetooth.h:	bswap_128(src, dst);
/usr/include/bluetooth/bluetooth.h:	bswap_128(src, dst);
```

dwaq, 2017-02-08:
On my raspberry pi 3, `bluetooth.h` does not include `bwap_128`. I have various other ***128 functions, so I've attached the file (renamed from bluetooth.h to bluetooth.txt to make github happy) so you can see which is the one that should be used.
[bluetooth.txt](https://github.com/labapart/gattlib/files/761998/bluetooth.txt)

oliviermartin, 2017-02-08:
Thanks @dwaq for the quick answer. Do you mind to test this pull-request https://github.com/labapart/gattlib/pull/9 ?
If it does not work I will have another way to fix the issue.

dwaq, 2017-02-08:
That worked! It build successfully and to be sure, I ran the `ble_scan` and got plausible results. I think that pull request can be merged and this issue closed. Thanks for your help throughout this process.

oliviermartin, 2017-02-08:
Thanks to you for testing and supporting this issue :-)

closed 2017-02-08 by oliviermartin