labapart/gattlib #7: Getting "fatal error: uuid.h: No such file or directory" when trying to use the library

noosxe, 2017-02-02:
Hi. I've successfully compiled the library. When trying to use it from my code I'm getting the following error:

/usr/local/include/gattlib.h:32:18: fatal error: uuid.h: No such file or directory
 #include "uuid.h"

Any idea why uuid.h is not being installed to system with gattlib ?

fcoldstar, 2017-02-05:
I got the same error.  Any idea how to fix it?

oliviermartin, 2017-02-06:
Hi all,
I have just committed a fix in this pull-request: https://github.com/labapart/gattlib/pull/8
I do not have my bluetooth setting to test it properly here (I should get it back in 2-3 days from now) but at least it seems to build. Let me know if it works for you.

oliviermartin, 2017-02-08:
I have just tested and made some fixes to my pull-request. It should now work. Do you mind to test it?

closed 2017-04-26 by oliviermartin