labapart/gattlib #10: multiple open-read-close causes: GLib-CRITICAL **: g_io_channel_unref: assertion 'channel != NULL' failed

stepen, 2017-02-16:
If I wrap the read_write block in read_write.c in a for-loop, re-compile and run a read the program very often crashes with a segmentation fault.

(process:14632): GLib-CRITICAL **: g_io_channel_unref: assertion 'channel != NULL' failed
Segmentation fault

'Wrapping' looks like (failure occurs only sometimes):

```
int n;
for (n=0; n<1000; n++)
{
  connection = gattlib_connect(NULL, argv[1], BDADDR_LE_PUBLIC, BT_SEC_LOW, 0, 0);
  if (connection == NULL) {
    fprintf(stderr, "Fail to connect to the bluetooth device.\n");
    return 1;
  }
  if (g_operation == READ) {
    len = gattlib_read_char_by_uuid(connection, &g_uuid, buffer, sizeof(buffer));
    printf("Read UUID completed: ");
    for (i = 0; i < len; i++)
      printf("%02x ", buffer[i]);
    printf("\n");
  } else {
    uint16_t handle = 0; //TODO: FIXME
    ret = gattlib_write_char_by_handle(connection, handle, buffer, sizeof(buffer));
    assert(ret == 0);
  }
  gattlib_disconnect(connection);
}
```

Latest gattlib and Bluetooth 5.43 on RaspberryPi3

oliviermartin, 2017-02-18:
Do you mind to try with the latest version of `gattlib`. I have just fixed the warning issue in https://github.com/labapart/gattlib/commit/ed3ff0627f8d60f54c2afd602bb8b68a8e1ca02b
I am wondering whether both issues are linked?

oliviermartin, 2017-02-18:
If the commit does not fix your issue you can have a look at the blog entry I just created: http://labapart.com/blogs/9-how-to-fix-glib-warnings-or-errors-with-gdb

It explains how to debug Glib error/warning/assertion using ``gdb.

stepen, 2017-02-18:
Just tried with the latest version of gattlib (as of 2017-02-18)  and re-run the tight for-loop as shown above. It seems that the 'assertion 'channel != NULL' failed' error is gone.  Fantastic!! Thanks!

FYI: I've reverted to the Raspberry PI's default BlueZ 5.23 (but the error was the same as before until your recent update).

closed 2017-02-19 by stepen