labapart/gattlib #12: Reading movement data

gracechristina, 2017-02-28:
I tried to use gattlib_read_char_by_uuid(connection, &g_uuid, buffer, sizeof(buffer)); and put it inside while loop to read data continously. However, it doesn't work. Is there any way to read data continuously upon request without using notification? Thank you for your help

oliviermartin, 2017-03-03:
Hi @gracechristina , what do you mean it does not work? I have just tried with the example `read_write` and it seems to work.with this code change:

```
index db03c74..83a7786 100644
--- a/examples/read_write/read_write.c
+++ b/examples/read_write/read_write.c
@@ -68,19 +68,21 @@ int main(int argc, char *argv[]) {
        if (g_operation == READ) {
-               len = gattlib_read_char_by_uuid(connection, &g_uuid, buffer, sizeof(buffer));
+               for (int counter = 0; counter < 10; counter++) {
+                       len = gattlib_read_char_by_uuid(connection, &g_uuid, buffer, sizeof(buffer));
 
-               printf("Read UUID completed: ");
-               for (i = 0; i < len; i++)
-                       printf("%02x ", buffer[i]);
-               printf("\n");
+                       printf("Read UUID completed: ");
+                       for (i = 0; i < len; i++)
+                               printf("%02x ", buffer[i]);
+                       printf("\n");
+               }
        } else {
                uint16_t handle = 0;
```

```
$ time ./examples/read_write/read_write DA:94:40:95:E0:87 read 0x2a38
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 
Read UUID completed: 03 

real	0m1.841s
user	0m1.832s
sys	0m0.004s
```

oliviermartin, 2017-03-03:
I have got the same result using `gattool` with the following bash program:

```
#!/bin/bash

for i in `seq 1 100`;
do
	gatttool -t random -b DA:94:40:95:E0:87 --char-read -a 0x000e
done 
```

closed 2017-04-26 by oliviermartin