labapart/gattlib #15: Error while making gattlib after couple of bluez version updates

irshad-qb, 2017-03-17:
I was able to build gattlib properly with bluez 5.43 version. Recently I played with bluez versions and tried 4.101 and couple of older versions by building the source file. Now I am not able to build the gattlib library. I am trying  
cmake .. && make install from my build folder.
I get the below error : 
``[  4%] Building C object bluez/CMakeFiles/gattlib.dir/gattlib_connect.c.o
/home/qburst/gattlib/bluez/gattlib_connect.c:165:22: error: unknown type name ‘BtIOSecLevel’
   uint8_t dest_type, BtIOSecLevel sec_level, int psm, int mtu,
                      ^
/home/qburst/gattlib/bluez/gattlib_connect.c:271:8: error: unknown type name ‘BtIOSecLevel’
 static BtIOSecLevel get_bt_io_sec_level(gattlib_bt_sec_level_t sec_level) {
        ^
/home/qburst/gattlib/bluez/gattlib_connect.c: In function ‘get_bt_io_sec_level’:
/home/qburst/gattlib/bluez/gattlib_connect.c:274:10: error: ‘BT_IO_SEC_SDP’ undeclared (first use in this function)
   return BT_IO_SEC_SDP;
          ^
/home/qburst/gattlib/bluez/gattlib_connect.c:274:10: note: each undeclared identifier is reported only once for each function it appears in
/home/qburst/gattlib/bluez/gattlib_connect.c:276:10: error: ‘BT_IO_SEC_LOW’ undeclared (first use in this function)
   return BT_IO_SEC_LOW;
          ^
/home/qburst/gattlib/bluez/gattlib_connect.c:278:10: error: ‘BT_IO_SEC_MEDIUM’ undeclared (first use in this function)
   return BT_IO_SEC_MEDIUM;
          ^
/home/qburst/gattlib/bluez/gattlib_connect.c:280:10: error: ‘BT_IO_SEC_HIGH’ undeclared (first use in this function)
   return BT_IO_SEC_HIGH;
          ^
/home/qburst/gattlib/bluez/gattlib_connect.c: In function ‘gattlib_connect_async’:
/home/qburst/gattlib/bluez/gattlib_connect.c:291:2: error: unknown type name ‘BtIOSecLevel’
  BtIOSecLevel bt_io_sec_level = get_bt_io_sec_level(sec_level);
  ^
/home/qburst/gattlib/bluez/gattlib_connect.c:293:9: warning: implicit declaration of function ‘initialize_gattlib_connection’ [-Wimplicit-function-declaration]
  return initialize_gattlib_connection(src, dst, dest_type, bt_io_sec_level,
         ^
/home/qburst/gattlib/bluez/gattlib_connect.c:293:9: warning: return makes pointer from integer without a cast [-Wint-conversion]
/home/qburst/gattlib/bluez/gattlib_connect.c: In function ‘gattlib_connect’:
/home/qburst/gattlib/bluez/gattlib_connect.c:316:2: error: unknown type name ‘BtIOSecLevel’
  BtIOSecLevel bt_io_sec_level = get_bt_io_sec_level(sec_level);
  ^
/home/qburst/gattlib/blue``

oliviermartin, 2017-03-17:
Hi @irshad-qb , I have just pushed a fix in https://github.com/labapart/gattlib/commit/0ee021f531bc29bb19eb53f3c03637a43feb3c47. Let me know if it works for you :-)

irshad-qb, 2017-03-17:
Thanks alot. Could you please explain the fix ? Was it with version 4.001 ? For me when I restarted after installing bluez 5.37 fixed the above issue. Also could you explain which version of bluez works best with this repo ? For me device scan fails to detect my BLE device in 5.37.

oliviermartin, 2019-07-15:
I am actively working with Bluez v5.48 (Bluez shipped with Ubuntu 18.04) and its seems to work fine. I close the issue for now. Feel free to open a new issue with new questions/issues.

closed 2019-07-15 by oliviermartin