labapart/gattlib #17: Unbearably slow connection

kristapsdreija, 2017-03-23:
Is it just me or is the connection time unbearably slow?
Using on custom nrf51 boards, but the ble_scan also connects super slowly to all other ble devices in range.
Any way how to speed things up? Tried both sync and async connections.

oliviermartin, 2017-03-23:
As you saw in the other threads, I cannot unfortunately reproduce the issue some (most?) people are seeing.

`ble_scan` has a timeout to wait for the bluetooth controller to scan and find advertising BLE devices around. Reducing this timeout to `0` will use BLE devices previously found.

At the moment, the timeout is set at `4` https://github.com/labapart/gattlib/blob/master/examples/ble_scan/ble_scan.c#L9.

The hardware I use to test: a 32bit x86 Ubuntu 16.04 laptop (the one I am using), a 64bit Ubuntu 14.04 laptop, nRF52 Development Kit (BLE peripheral for `gattlib`), a 96board Hikey board (BLE peripheral for `gattlib`). The performance looks similar to mainstream `gatttool` on all devices.

kristapsdreija, 2017-03-23:
The scanning is fast, the connection time is super slow! Connect->read/write->disconnect

Buguito, 2017-06-28:
@kristapsdreija Don't know if you ever fixed this. But i think the slowness is because of a TIMEOUT set to 4 also in the gattlib.c file. And that timeout is used at the connect method.

It's defined as: #define CONNECT_TIMEOUT  4 in gattlib.c

Changing that to 0, will make it superfast. I'm experiencing another issue when polling multiple BLE devices and trying to read. On reads just fine, the other one starts ok and then fails (it kind of disconnects in the middle when trying to read it), i get a Bluez error "not connected", but it actually should be connected since i have a LED in the device that tells me so.

This is not related to the CONNECT_TIMEOUT tough.

Regards, Alan