labapart/gattlib #18: cmake throws errors

wurstfingerzwerg, 2017-03-23:
Following your readme while trying to make the gattlib under a debian jessie throws the following errors:

pi@raspberrypi ~/gattlib/build $ cmake -DGATTLIB_FORCE_DBUS=TRUE ..
Build DBus gattlib for Bluez v5.23
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/gattlib/build
pi@raspberrypi ~/gattlib/build $ make
[  4%] Generate D-Bus 'org.bluez.GattService1.xml'
[  8%] Generate D-Bus 'org.bluez.Adapter1.xml'
[ 12%] Generate D-Bus 'org.bluez.Device1.xml'
[ 16%] Generate D-Bus 'org.bluez.GattCharacteristic1.xml'
[ 20%] Generate D-Bus 'org.bluez.GattDescriptor1.xml'
Scanning dependencies of target gattlib
[ 25%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib.c.o
[ 29%] Building C object dbus/CMakeFiles/gattlib.dir/bluez5/lib/uuid.c.o
/home/pi/gattlib/dbus/bluez5/lib/uuid.c: In function ‘bt_uuid_to_le’:
/home/pi/gattlib/dbus/bluez5/lib/uuid.c:306:3: warning: implicit declaration of function ‘bswap_128’ [-Wimplicit-function-declaration]
   bswap_128(&src->value.u128, dst);
   ^
[ 33%] Building C object dbus/CMakeFiles/gattlib.dir/__/gattlib_common.c.o
[ 37%] Building C object dbus/CMakeFiles/gattlib.dir/org-bluez-adaptater1.c.o
[ 41%] Building C object dbus/CMakeFiles/gattlib.dir/org-bluez-device1.c.o
[ 45%] Building C object dbus/CMakeFiles/gattlib.dir/org-bluez-gattcharacteristic1.c.o
[ 50%] Building C object dbus/CMakeFiles/gattlib.dir/org-bluez-gattdescriptor1.c.o
[ 54%] Building C object dbus/CMakeFiles/gattlib.dir/org-bluez-gattservice1.c.o
[ 58%] Linking C shared library libgattlib.so
[ 58%] Built target gattlib
[ 62%] Building C object examples/ble_scan/CMakeFiles/ble_scan.dir/ble_scan.c.o
[ 66%] Linking C executable ble_scan
../../dbus/libgattlib.so: undefined reference to `bswap_128'
collect2: error: ld returned 1 exit status
examples/ble_scan/CMakeFiles/ble_scan.dir/build.make:95: recipe for target 'examples/ble_scan/ble_scan' failed
make[2]: *** [examples/ble_scan/ble_scan] Error 1
CMakeFiles/Makefile2:140: recipe for target 'examples/ble_scan/CMakeFiles/ble_scan.dir/all' failed
make[1]: *** [examples/ble_scan/CMakeFiles/ble_scan.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

The line "Build DBus gattlib for Bluez v5.23" is also strange, whilst a Bluez 5.41 is installed on that pi.

oliviermartin, 2017-04-26:
Sorry, I missed your issue.

Interesting about Bluez v5.23. How have you installed Bluez? Using the distribution package or manually?

wurstfingerzwerg, 2017-04-26:
I installed Bluez 5.41 manually. bluetoothd -v returns 5.41

nojanen, 2017-04-28:
I had the same problem, and I got it solved by removing package `libbluetooth-dev` and adding option `--enable-library` to BlueZ compilation configurations. I.e.

```
sudo apt-get purge libbluetooth-dev
cd <bluez src>
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-experimental --with-systemdsystemunitdir=/lib/systemd/system --with-systemduserunitdir=/usr/lib/systemd --enable-library
make
sudo checkinstall
```

oliviermartin, 2017-07-02:
Thanks a lot @nojanen ! @wurstfingerzwerg I will close the issue for now, feel free to re-open it if the issue persists.

closed 2017-07-02 by oliviermartin