labapart/gattlib #19: does mtu got a support?

jasonbu, 2017-04-26:


oliviermartin, 2017-04-26:
Did you expect a larger MTU in your case? Have you tried your data notification with `gatttool`?

I had a quick look to gattlib code and I cannot see hard-coded size, but I might have missed it.

jasonbu, 2017-04-26:
I'm trying to use gatttool is this repo, it look not easy to use. And now found the gatttool within bluez, trying.

the deivce it with Android and windows tested. it should not be a trouble with device.

jasonbu, 2017-04-26:
using bluez gatttool seems also got not action for mtu. with device logcat i got a success, but host can only get data for length 20 bytes.

Idorobots, 2018-07-29:
I'm running into the same issue. Trying to connect to a BLE device using gattlib I'm getting notifications limited to max 20 bytes of data, while the same device sends 58 bytes when using WebBluetooth to connect to it. This happens regardless of wether I'm using the DBUS version or the legacy Bluez one even if I set the MTU when connecting to the device. Any clues what may be going wrong?

Idorobots, 2018-07-29:
I think I found a workaround for this issue (the issue being `gatt_connect()` supplied MTU setting being ignored), based on the gattool.c example:

```c
const uint16_t opt_mtu = 128; // Your desired MTU size.

// FIXME This is an internal thing that shouldn't be accesed from outside.
typedef struct {
  GIOChannel* io;
  GAttrib* attrib;
  void* characteristics;
  int characteristic_count;
} gattlib_context_t;

static void exchange_mtu_cb(guint8 status, const guint8 *pdu, guint16 plen, gpointer user_data) {
  gattlib_context_t* conn_context = (gattlib_context_t*) user_data;
  uint16_t mtu;
  dec_mtu_resp(pdu, plen, &mtu);

  mtu = MIN(mtu, opt_mtu);

  if (g_attrib_set_mtu(conn_context->attrib, mtu))
    printf("MTU was exchanged successfully: %d\n", mtu);
  else
    printf("Error exchanging MTU\n");
}

// Elswhere

gatt_connection_t *connection = gattlib_connect(NULL, addr, BDADDR_LE_PUBLIC, BT_SEC_MEDIUM, 0, 0);
gatt_exchange_mtu(conn_context->attrib, opt_mtu, exchange_mtu_cb, connection->context);
```

Obviously this isn't ideal and it requires the legacy Bluez version instead of the Dbus one, but at least this makes it possible to retrieve full updates from the devices.

oliviermartin, 2019-07-15:
I have just introduced gattlib Streaming API. When opening a stream on a GATT characteristic you get the MTU back (see: https://github.com/labapart/gattlib/blob/master/include/gattlib.h#L395). As far I can see I manage to get a large MTU (in my case was 273 bytes - my BLE peripheral limitation).

I am assuming this ticket as fixed (at least until someone manage to duplicate the issue and fix it as it seems to work for me).

closed 2019-07-15 by oliviermartin