labapart/gattlib #20: examples/gatttool/gatttool coredumps with Bluez 5.23

wojciiii, 2017-05-25:
The following command segfaults:
`#  ./examples/gatttool/gatttool -b F7:35:49:55:35:E5 -I -t random`

This is using Bluez 5.23.

The result of backtrace in gdb:
`#0  0x000137e0 in cmd_connect (argcp=1, argvp=0xeeb7c0) at /home/wojci/remote-git/gattlib/examples/gatttool/interactive.c:311
#1  0x000149e8 in parse_line (line_read=0xee6170 "connect") at /home/wojci/remote-git/gattlib/examples/gatttool/interactive.c:795
#2  0x76d28b30 in rl_callback_read_char () from /lib/arm-linux-gnueabihf/libreadline.so.6
#3  0x00014a58 in prompt_read (chan=0xed9b00, cond=G_IO_IN, user_data=0x0) at /home/wojci/remote-git/gattlib/examples/gatttool/interactive.c:810
#4  0x76d8c1c4 in g_main_dispatch (context=0xed9a80) at /build/glib2.0-tTvduh/glib2.0-2.42.1/./glib/gmain.c:3111
#5  g_main_context_dispatch (context=context@entry=0xed9a80) at /build/glib2.0-tTvduh/glib2.0-2.42.1/./glib/gmain.c:3710
#6  0x76d8c588 in g_main_context_iterate (context=0xed9a80, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at /build/glib2.0-tTvduh/glib2.0-2.42.
1/./glib/gmain.c:3781
#7  0x76d8c8ec in g_main_loop_run (loop=0xed97e8) at /build/glib2.0-tTvduh/glib2.0-2.42.1/./glib/gmain.c:3975
#8  0x00014cb8 in interactive (src=0x0, dst=0xed9760 "F7:35:49:55:35:E5", dst_type=0xedb8c8 "random", psm=0) at /home/wojci/remote-git/gattlib/examples/gatttool/interac
tive.c:867
#9  0x00012c94 in main (argc=1, argv=0x7e8157a4) at /home/wojci/remote-git/gattlib/examples/gatttool/gatttool.c:525`

wojciiii, 2017-05-25:
[core.gz](https://github.com/labapart/gattlib/files/1029025/core.gz)

twonav, 2017-06-02:
Also with bluez 4.101

oliviermartin, 2017-06-02:
@wojciiii Do you mind to attach the binary you have been using? I will need it to match your `core.gz`.
I cannot duplicate the issue on my computer.

wojciiii, 2017-06-02:
Hi,

I don't have that any more. :(

I don't think that this is a problem with the library, as it works as
expected in my own software.

oliviermartin, 2017-07-02:
Thanks @wojciiii  for the update. I will close the issue in this case.

closed 2017-07-02 by oliviermartin