labapart/gattlib #28: read_write example returning -1 at gattlib_read_char_by_uuid in Bluez 5.45

Buguito, 2017-06-27:
Hi Olivier and dear devs.. I'm porting a C++ program of my own that was properly working with a RFDuino device also of my own. We were using QTBluetooth, and switched to pure C, so i went ahead and started using this library gattlib for future replacement of RFDuino with some NRF module and do an embedded system.

The LE Scan works just fine, and also the provided example for characteristics discovery (discovery). But when i try to read the value from my LE device (an RFDuino that reads a temp and humidity sensor), i'm getting a ret = -1 from the gattlib_read_char_by_uuid call at read_write.c example.

So i went ahead and tried to trace where the issue was, putting some printf in the gattlib.c at the dbus/ folder and recompiling the examples.

The issue is at org_bluez_gatt_characteristic1_call_read_value_sync(characteristic, &out_value, NULL, &error)

Using the original gatttool i can read properly (don't know if gatttool actually uses DBUS API or not...)

Here are the outputs:

root@gizmo:~/gattlib/build/examples/discover# ./discover D5:A9:88:3E:CA:09
service[0] start_handle:00 end_handle:00 uuid:0x2220
service[1] start_handle:00 end_handle:00 uuid:0x1801
characteristic[0] properties:12 value_handle:0000 uuid:0x2221
characteristic[1] properties:0c value_handle:0000 uuid:0x2223
characteristic[2] properties:2f value_handle:0000 uuid:0x2222
characteristic[3] properties:20 value_handle:0000 uuid:0x2a05

root@gizmo:~/gattlib/build/examples/read_write# ./read_write D5:A9:88:3E:CA:09 read 00002221-0000-1000-8000-00805f9b34fb
Connecting..
Connected.
UUID: 00002221-0000-1000-8000-00805f9b34fb
ERROR AT READ_VALUE_SYNC
Error: **GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "ReadValue" with signature "" on interface "org.bluez.GattCharacteristic1" doesn't exist**
Ret: -1

Now when using gatttool:

root@gizmo:~/gattlib/build/examples/read_write# gatttool -t random -b D5:A9:88:3E:CA:09 --char-read -a 0x000e
Characteristic value/descriptor: 49 00 00 88 42 33 33 bb 41 00 00 00 00 00 00 62 00 00 30 00

It just reads the characteristic just fine (20 bytes) as intended.

Any thoughts on the "GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "ReadValue" with signature "" on interface "org.bluez.GattCharacteristic1", error?

It's driving me insane.

Thank you!

Buguito, 2017-06-27:
Some extra info. getting the primary services with gatttool:

root@gizmo:~/gattlib/build/examples/read_write# gatttool -t random -b D5:A9:88:3E:CA:09 --primary
attr handle = 0x0001, end grp handle = 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0008, end grp handle = 0x000b uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x000c, end grp handle = 0xffff uuid: 00002220-0000-1000-8000-00805f9b34fb

As you can see, there's a 0x000c which is the only service available for read/write at RFDuino propietary GATT Profile.

And if i evesdrop on dbus with dbus-monitor --system while using your read_write.c example, i have this:

method call sender=:1.49 -> dest=org.freedesktop.DBus serial=19 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RemoveMatch
   string "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged',path='/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service0008/char0009',arg0='org.bluez.GattCharacteristic1'"
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=20 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RemoveMatch
   string "type='signal',sender='org.bluez',interface='org.bluez.GattCharacteristic1',path='/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service0008/char0009'"
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=21 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged',path='/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service000c/char000d',arg0='org.bluez.GattCharacteristic1'"
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=22 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender='org.bluez',interface='org.bluez.GattCharacteristic1',path='/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service000c/char000d'"
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=23 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=StartServiceByName
   string "org.bluez"
   uint32 0
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=24 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=GetNameOwner
   string "org.bluez"
method call sender=:1.49 -> dest=:1.2 serial=25 path=/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service000c/char000d; interface=org.freedesktop.DBus.Properties; member=GetAll
   string "org.bluez.GattCharacteristic1"
method return sender=:1.2 -> dest=:1.49 reply_serial=25
   array [
      dict entry(
         string "UUID"
         variant             string "00002221-0000-1000-8000-00805f9b34fb"
      )
      dict entry(
         string "Service"
         variant             object path "/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service000c"
      )
      dict entry(
         string "Value"
         variant             array [
            ]
      )
      dict entry(
         string "Notifying"
         variant             boolean false
      )
      dict entry(
         string "Flags"
         variant             array [
               string "read"
               string "notify"
            ]
      )
   ]
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=26 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RemoveMatch
   string "type='signal',sender=':1.2'"
method call sender=:1.49 -> dest=org.freedesktop.DBus serial=27 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RemoveMatch
   string "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.ObjectManager',path='/'"
method call sender=:1.49 -> dest=:1.2 serial=28 path=/org/bluez/hci0/dev_D5_A9_88_3E_CA_09/service000c/char000d; interface=org.bluez.GattCharacteristic1; member=ReadValue
error sender=:1.2 -> dest=:1.49 error_name=org.freedesktop.DBus.Error.UnknownMethod reply_serial=28
   string "Method "ReadValue" with signature "" on interface "org.bluez.GattCharacteristic1" doesn't exist
"
method call sender=:1.49 -> dest=:1.2 serial=29 path=/org/bluez/hci0/dev_D5_A9_88_3E_CA_09; interface=org.bluez.Device1; member=Disconnect
signal sender=:1.2 -> dest=(null destination) serial=193 path=/org/bluez/hci0/dev_D5_A9_88_3E_CA_09; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Device1"
   array [
      dict entry(
         string "ServicesResolved"
         variant             boolean false
      )
   ]
   array [
   ]
method return sender=:1.2 -> dest=:1.49 reply_serial=29
signal sender=:1.2 -> dest=(null destination) serial=195 path=/org/bluez/hci0/dev_D5_A9_88_3E_CA_09; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Device1"
   array [
      dict entry(
         string "Connected"
         variant             boolean false
      )
   ]
   array [
   ]
signal sender=org.freedesktop.DBus -> dest=(null destination) serial=62 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.49"
   string ":1.49"
   string ""
signal sender=:1.0 -> dest=(null destination) serial=312 path=/org/freedesktop/systemd1; interface=org.freedesktop.systemd1.Manager; member=UnitRemoved
   string "sys-subsystem-bluetooth-devices-hci0:69.device"
   object path "/org/freedesktop/systemd1/unit/sys_2dsubsystem_2dbluetooth_2ddevices_2dhci0_3a69_2edevice"
signal sender=:1.0 -> dest=(null destination) serial=313 path=/org/freedesktop/systemd1; interface=org.freedesktop.systemd1.Manager; member=UnitRemoved
   string "sys-devices-pci0000:00-0000:00:11.0-0000:02:00.0-usb2-2\x2d2-2\x2d2.1-2\x2d2.1:1.0-bluetooth-hci0-hci0:69.device"
   object path "/org/freedesktop/systemd1/unit/sys_2ddevices_2dpci0000_3a00_2d0000_3a00_3a11_2e0_2d0000_3a02_3a00_2e0_2dusb2_2d2_5cx2d2_2d2_5cx2d2_2e1_2d2_5cx2d2_2e1_3a1_2e0_2dbluetooth_2dhci0_2dhci0_3a69_2edevice"

There's a service000c in the debug output, don't know if that means anything to you.

Buguito, 2017-06-27:
Characteristics with gatttool:

root@gizmo:~/gattlib/build/examples/read_write# gatttool -t random -b D5:A9:88:3E:CA:09 --characteristics
handle = 0x0002, char properties = 0x0a, char value handle = 0x0003, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0004, char properties = 0x02, char value handle = 0x0005, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0006, char properties = 0x02, char value handle = 0x0007, uuid = 00002a04-0000-1000-8000-00805f9b34fb
handle = 0x0009, char properties = 0x20, char value handle = 0x000a, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x000d, char properties = 0x12, char value handle = 0x000e, uuid = 00002221-0000-1000-8000-00805f9b34fb
handle = 0x0010, char properties = 0x0c, char value handle = 0x0011, uuid = 00002222-0000-1000-8000-00805f9b34fb
handle = 0x0012, char properties = 0x0c, char value handle = 0x0013, uuid = 00002223-0000-1000-8000-00805f9b34fb

Buguito, 2017-06-27:
Ok, i've installed bluez-5.41, recompiled so no D-BUS API is used, changed the ADDR type to "random" in the examples and it's working now. My device might have been incompatible with the DBUS-API or i failed to change the ADDR type to random previously in the examples but not sure about the latter. Mostly my device was not compatible since it's a RFDuino with old firmware.

Buguito, 2017-06-27:
Ok, so i wanted to retry with D-BUS API. From what i understand the problem is that the signature for the call to ReadValue method on the org.bluez.GattCharacteristic1 is wrong. It's "empty", while using dfeet i send for {} as input parameter and i get the data correctly.

It expects dict of {string, variant}. Any hints on how to fix the signature for that method?. I see that on the .xml file for the GattCharacteristic, there's a line commented out:

<arg name="options" type="a{sv}" direction="in"/>

If i uncomment that, well it won't compile, but i think i'm going in the right direction?. If the parameters are not defined for the call and i'm getting an empty signature, where is it defined?.

Regards, Alan

Buguito, 2017-06-27:
Ok i fixed it. Olivier in case you want to change this in future releases to comply with the latest signatures for ReadValue and WriteValue, this as basically the changes i had to make.

First remove the commented lines in the XML's for gdbus-codegen, so there's an input parameter of type="a{sv}".

Then in gattlib.c at /dbus, add the parameter in both read_value and write_value as follows:

GVariant *in_params = g_variant_parse(G_VARIANT_TYPE_VARDICT, "{}", NULL, NULL, NULL);
org_bluez_gatt_characteristic1_call_read_value_sync(characteristic, in_params, &out_value, NULL, &error);

That must be added in both read_value calls (there's a async and one withouth it), and also in the write_value corresponding functions. Have not tested the write yet, but the read operation now is working swell in via DBUS API running Bluez 5.45.

Regards, Alan

oliviermartin, 2017-07-02:
Thanks a lot @Buguito for the investigation. I created a fix following your comment in this commit c890f64dd9. I have not tested it yet - but it builds... I hope to test it myself soon.

I close the issue for now. Feel free to re-open it.

closed 2017-07-02 by oliviermartin