labapart/gattlib #29: Library is not complaint with multi devices with same GATT Profile

Buguito, 2017-06-28:
I'm working with several devices that are exactly the same so they have the same profiles and same services and characteristics.

Unless i'm wrong, your static OrgBluezGattCharacteristic1 *get_characteristic_from_uuid(const uuid_t* uuid) at gattlib.c, just loops through the characteristics on the DBUS and picks the one that conforms to the handle passed, for example to read_char.c example.

So in my case even i have to devices with mac address: D5:A9:88:3E:CA:09 and D7:B9:CE:79:AA:FA, both have a same characteristic 0x2221 for certain service.

I've printf'ed the moment you decide that the method that gets the characteristic should returned in static OrgBluezGattCharacteristic1 *get_characteristic_from_uuid(const uuid_t* uuid)

To be more specific, here: 

`if (gattlib_uuid_cmp(uuid, &characteristic_uuid) == 0) {
                                printf("PATH: %s\n",object_path);
                                break;
                        }
`

And what happens is that the object_path for the device I'M NOT CONNECTED TO is returned, instead of the object_path of the very same characteristic for the one I'M connected to. As a result the wrong characteristic is returned and the DBUS API call is made on the wrong device.

root@gizmo:~/gattlib/build/examples/read_write# ./read_write **D7:B9:CE:79:AA:FA** read 0x2221
Connecting..
Connected.
UUID: 0x2221
PATH: /org/bluez/hci0/**dev_D5_A9_88_3E_CA_09**/service000c/char000d
Error: GDBus.Error:org.bluez.Error.Failed: Failed to send read request
Ret: -1

Looking forward to your reply.

Regards, Alan

oliviermartin, 2017-07-02:
Thanks Alan for reporting the issue. I have not investigated it yet but your analysis seems to be correct.

I do not know when I will be able to fix it myself. But if you decide to send a pull-request with a fix I will be happy to review it and merge it :-)

PhilippFinke, 2019-01-29:
This seems still to be an issue. I'm currently having the same problems.