labapart/gattlib #40: Scan and discover work just fine, but cannot read or write.

slashnull, 2017-08-14:
CentOS7
bluez-5.41-1.el7
3.10.0-514.26.2.el7.x86_64

git clone https://github.com/labapart/gattlib.git
cd gattlib
mkdir build && cd build
cmake -DGATTLIB_FORCE_DBUS=TRUE -DCMAKE_BUILD_TYPE=Debug ..
make

```
[user@Series2s10 build]$ sudo ./examples/discover/discover D8:46:CC:8B:F2:A1
service[0] start_handle:00 end_handle:00 uuid:0xff20
service[1] start_handle:00 end_handle:00 uuid:0x1801
service[2] start_handle:00 end_handle:00 uuid:0xfee7
characteristic[0] properties:10 value_handle:0000 uuid_str:0xff22 uuid:6
characteristic[1] properties:04 value_handle:0000 uuid_str:0xff21 uuid:6
characteristic[2] properties:02 value_handle:0000 uuid_str:0xfec9 uuid:6
characteristic[3] properties:37 value_handle:0000 uuid_str:0xfea1 uuid:6
[user@Series2s10 build]$ sudo ./examples/read_write/read_write D8:46:CC:8B:F2:A1 read 0xff21

(process:9375): GLib-CRITICAL **: '{}' is not a valid GVariant format string

(process:9375): GLib-CRITICAL **: g_variant_new: assertion 'valid_format_string (format_string, TRUE, NULL) && format_string[0] != '?' && format_string[0] != '@' && format_string[0] != '*' && format_string[0] != 'r'' failed

(process:9375): GLib-CRITICAL **: g_variant_get_type: assertion 'value != NULL' failed

(process:9375): GLib-CRITICAL **: g_variant_type_is_subtype_of: assertion 'g_variant_type_check (type)' failed

(process:9375): GLib-CRITICAL **: g_variant_get_type_string: assertion 'value != NULL' failed

(process:9375): GLib-ERROR **: g_variant_new: expected GVariant of type 'a{sv}' but received value has type '(null)'
/var/tmp/scl0KUnBO: line 8:  9375 Trace/breakpoint trap   (core dumped) './examples/read_write/read_write' 'D8:46:CC:8B:F2:A1' 'read' '0xff21'

```
Any clue on what I could do to trouble shoot this further?  I thinking perhaps there is a newer version of Glib that is expected?

Thanks.

mlaidma, 2017-08-29:
I am experiencing the same problem when trying to gattlib_read_char_by_uuid(). Installed gattlib as in the readme, without -DGATTLIB_FORCE_DBUS=TRUE. I am running Raspbian Stretch (Linux raspberrypi 4.9.41-v7+) and bluez 5.43. 

Have you found a solution?

c-larsen, 2017-09-08:
This sounds like the problem I have created a fix for. See https://github.com/labapart/gattlib/pull/45 and check if that solves the issue.

oliviermartin, 2017-09-08:
Thanks a lot @c-larsen for fixing the issue. I have just merged it.

closed 2017-09-08 by oliviermartin

c-larsen, 2017-09-21:
Please see https://github.com/labapart/gattlib/pull/46 for fixing the write functionality. I discovered that the arguments was given in the wrong order in the code. I can have a look at the errors as well.