labapart/gattlib #42: Peripheral BLE device

AnDrOiD-73, 2017-09-06:
Thank you for your library.
Can I use your library for create peripheral BLE device?
I need the following functional:
add custom service and set UUID for it
add custom characteristics to this service and set UUID for each of them
read only characteristic
write only characteristic
notification characteristic
change MTU by central device
notify connected devices when value has been changed
callback for:
connect/disconnect device
read characteristic
write characteristic

phishstang65, 2017-09-09:
I agree - this would be an Awesome addition to this library. I don't think it currently allows you to create a GATT Server. BlueZ 4.101 source code has "tools/btgatt-server.c".

oliviermartin, 2017-10-07:
I have already created a couple of GATT server (ie: Peripheral BLE device) using Bluez DBus API. And I think it is the best approach rather than to create a library.