labapart/gattlib #43: gattlib_write_char_by_uuid() max length?

zevv, 2017-09-07:
I'm probably doing something wrong, but I'm not able to send more then 8 bytes of data using gattlib_write_char_by_uuid(), anything longer gets discarded although the function just returns 0 as if everything is ok.

Any ideas how to debug this appreciated.

oliviermartin, 2017-10-07:
Hi @zevv , the best way to debug it is to test with bluez `gatttool`. For instance: `char-write-cmd 0x0028 0x123456789ABCDEF` (where `0x0028` would be the handle of one of your GATT characteristics and `0x123456789ABCDEF` the value to write).

But I suspect 8 bytes to be the MTU of your target device.

enmanoj, 2018-03-01:
This is because the data length is misinterpreted in the below function (bluez/gattlib_read_write.c)
`gattlib_write_char_by_uuid(gatt_connection_t* connection, uuid_t* uuid, const void* buffer, size_t buffer_len)`
here `return gattlib_write_char_by_handle(connection, handle, buffer, sizeof(buffer));`
the return statement's last parameter is sizeof(buffer), which always gives 4 bytes(size of pointer) as buffer is of type void*.
This can be solved by changing the parameter to "buffer_len" taken by gattlib_write_char_by_uuid as input parameter, ie. the return statement should be 
`return gattlib_write_char_by_handle(connection, handle, buffer, buffer_len);`
Doing this and recompiling and installing will fix this

oliviermartin, 2019-03-05:
Fix in https://github.com/labapart/gattlib/commit/7ad64704cb83660f5ae9b2d6aa7ab3541bf8760d

closed 2019-03-05 by oliviermartin