labapart/gattlib #55: make command throws a error

joswa, 2018-01-03:
pi@raspberrypi:~/gattlib $ make
[  4%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib.c.o
In file included from /usr/lib/gcc/arm-linux-gnueabihf/4.9/include-fixed/syslimits.h:7:0,
                 from /usr/lib/gcc/arm-linux-gnueabihf/4.9/include-fixed/limits.h:34,
                 from /usr/lib/arm-linux-gnueabihf/glib-2.0/include/glibconfig.h:11,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /home/pi/gattlib/dbus/gattlib.c:24:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include-fixed/limits.h:168:61: error: no include path in which to search for limits.h
 #include_next <limits.h>  /* recurse down to the real one */
                                                             ^
In file included from /usr/include/glib-2.0/glib/galloca.h:32:0,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /home/pi/gattlib/dbus/gattlib.c:24:
/usr/include/glib-2.0/glib/gtypes.h:35:18: fatal error: time.h: No such file or directory
 #include <time.h>
                  ^
compilation terminated.
dbus/CMakeFiles/gattlib.dir/build.make:82: recipe for target 'dbus/CMakeFiles/gattlib.dir/gattlib.c.o' failed
make[2]: *** [dbus/CMakeFiles/gattlib.dir/gattlib.c.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'dbus/CMakeFiles/gattlib.dir/all' failed
make[1]: *** [dbus/CMakeFiles/gattlib.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

oliviermartin, 2018-01-03:
Have you installed `libc6-dev` package of your distribution? Which distribution are you using?

I close the issue at the moment, fill free to re-open it after confirming that `libc6-dev` is installed.

closed 2018-01-03 by oliviermartin

joswa, 2018-01-06:
Yes i installed libc-dev package still there is issue while giving make command.

**Distribution : Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux**

PTmv, 2018-04-17:
When trying to build gattlib  for Raspbian for Raspberry Pi 3 on Ubuntu v 16.04.2, having installed BlueZ v 5.49 and Linaro cross-compiler v 4.9, I am getting the following error:

$ make
[  1%] Building C object bluez/CMakeFiles/gattlib.dir/gattlib_adapter.c.o
In file included from /home/ptomov/Toolchains/gcc-linaro-4.9-2015.05-x86_64_arm-linux-gnueabihf/lib/gcc/arm-linux-gnueabihf/4.9.3/include-fixed/syslimits.h:7:0,
                 from /home/ptomov/Toolchains/gcc-linaro-4.9-2015.05-x86_64_arm-linux-gnueabihf/lib/gcc/arm-linux-gnueabihf/4.9.3/include-fixed/limits.h:34,
                 from /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:11,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /home/ptomov/Downloads/GATTLIB/gattlib-master/bluez/gattlib_internal.h:27,
                 from /home/ptomov/Downloads/GATTLIB/gattlib-master/bluez/gattlib_adapter.c:1:
/home/ptomov/Toolchains/gcc-linaro-4.9-2015.05-x86_64_arm-linux-gnueabihf/lib/gcc/arm-linux-gnueabihf/4.9.3/include-fixed/limits.h:168:61: fatal error: limits.h: No such file or directory
 #include_next <limits.h>  /* recurse down to the real one */
                                                             ^
compilation terminated.
bluez/CMakeFiles/gattlib.dir/build.make:62: recipe for target 'bluez/CMakeFiles/gattlib.dir/gattlib_adapter.c.o' failed
make[2]: *** [bluez/CMakeFiles/gattlib.dir/gattlib_adapter.c.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'bluez/CMakeFiles/gattlib.dir/all' failed
make[1]: *** [bluez/CMakeFiles/gattlib.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Please advise me how to solve this problem. Thanks.