labapart/gattlib #56: read_write example, Read characteristics by UUID failed.

rarok59, 2018-01-06:
Hello all,

I have a problem with the read_write example program. I am using rpi3 with installed Bluez v5.42. During installation I was following the instructions from here: [https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation](url). 

I want to use this example as a starting point for application which will be able to read data continuously from my BLE device. While trying to use this example an error occures: "Read characteristics by UUID failed: Internal application error: I/O". I have checked if I am able to read this characteristic using gatttool command "char-read-uuid" and there is the same error.  Additionally I can say that using gatttool command "char-read-hnd" I am able to read this value without any problem, but I am struggling to do this by uuid. Does anyone knows a solution for this? Maybe someone has similiar problem as I have? Is there any way to change code in example to use gatttool char-read-hnd?

P.S. I am able to read data from my BLE device using other devices so there is no way its a problem with my BLE device.
Kind regards,
Emil

oliviermartin, 2018-01-18:
Out of curiosity, which BLE device are you trying to access? In case people have similar issue with this device.

Do you confirm you use `gattlib` with DBUS support?

rarok59, 2018-01-22:
Hello,

Sorry for late response. The BLE device is programable module from Silicon Labs. It's BGM111 which supports BLE.

I am not sure if You are asking for flag "-DGATTLIB_FORCE_DBUS=TRUE"  while I am performing cmake?

Kind regards,
Emil