labapart/gattlib #57: read_write example

max2323A, 2018-01-07:
Hello everyone,
I'm new in gattlib, I have problem with read_write example different from other issue (I think).
I'm trying to get my raspberry zero w to communicate with an Arduino BLE device.
with gatttool everything works fine, but with this example it gives me an error.

gatttool everything good:
[C3:03:00:CA:0C:62][LE]> char-read-uuid 00002902-0000-1000-8000-00805f9b34fb
handle: 0x000e   value: 00 00
[C3:03:00:CA:0C:62][LE]> char-read-uuid 2902
handle: 0x000e   value: 00 00

read_write example give me error:
pi@raspberrypi:~/gattlib/examples/read_write $ ./read_write C3:03:00:CA:0C:62 read 00002902-0000-1000-8000-00805f9b34fb
read_write: /home/pi/gattlib/examples/read_write/read_write.c:81: main: Assertion `ret == 0' failed.
Annullato

pi@raspberrypi:~/gattlib/examples/read_write $ ./read_write C3:03:00:CA:0C:62 read 2902
read_write: /home/pi/gattlib/examples/read_write/read_write.c:81: main: Assertion `ret == 0' failed.
Annullato

I tried also the ble_scan example and it works well
(bluez-5.48)

best regard
Davide

oliviermartin, 2018-01-18:
Try to print the `ret` value to understand a bit better the error message.  
You could also add `printf()` inside `gattlib_read_char_by_uuid()` to get a better idea where it fails.

oliviermartin, 2019-03-05:
@max2323A Bluez v5.48 does not allow to access CCC (Client Characteristic Configuration) descriptor through the DBUS interface (interface used by gattlib for Bluez v5.42).
Normally you should not need to access it except for debugging purposes.

Any use case in mind?

I have added an error message in gattlib to signal this limitation: https://github.com/labapart/gattlib/commit/c2e48c3fd69c93e2ab5ece7d3cd0fe8fe7449e15

I close your issue. Feel free to re-open it if you think it is relevant.

closed 2019-03-05 by oliviermartin