labapart/gattlib #59: gattlib_write_char_by_handle ??

DrEdHepler, 2018-01-17:
I'm attempting to use "gattlib_write_char_by_handle" to enable notification for my device (as suggested in README.md) and couldn't get it to work.  (I can get it to work when writing to handles using gatttool.)   So I looked in gattlib.c and found that this function just returns -1!!

Is this function no longer supported?

Also, using the stock gatttool to show characteristics list the appropriate handle for each one.   But when using the ble_scan example with my device, all handles are reported as 0.      Are handles not supported in gattlib?

Thanks!

DrEdHepler, 2018-01-17:
Looks like descriptors/attributes are not handled by gattlib?
And it looks like gattlib_notification_start() replaces the "more traditional method" of writing the value of 0x0001 to the handle of the config attributes of the characteristic that is to be sent as the notification data.   Traditionally/Usually this is the attribute/descriptor adjacent to the data (being sent in the notification), so has a handle of +1 relative to the data handle.

Does gattlib_notification_start() do this (write of 0x0001)??

With my BLE device, I can use gatttool (the one that is delivered as part of bluez) to write 0x0001 to the appropriate handle and see the notifications arrive.    However, when calling gattlib_notification_start()   (with the appropriate arguments),  the notifications never start.     I've looked through the source code, and don't see where the 0x0001 might be written.    And the ability to use gattlib_write_char_by_handle() no longer seems to be available.

Suggestions?   What am I doing wrong?

Thanks!

oliviermartin, 2018-01-18:
Which version of Bluez are you using?

Yes, `gattlib_notification_start()` does this (ie: write 1 to the appropriate handle). See https://github.com/labapart/gattlib/blob/0ee021f531bc29bb19eb53f3c03637a43feb3c47/bluez/gattlib_read_write.c#L200 for legacy Bluez. But I confirm the new DBUS Bluez also write 1.
I have not checked but maybe DBUS Bluez does not allow to write to the adjacent attribute/descriptor.

oliviermartin, 2018-01-18:
@DrEdHepler I tried to reply the emails you recently sent me. But I received: 

```
Undelivered Mail Returned to Sender
This is the mail system at host mo7.mail-out.ovh.net.

I'm sorry to have to inform you that your message could not
be delivered to one or more recipients. It's attached below.

For further assistance, please send mail to postmaster.

If you do so, please include this problem report. You can
delete your own text from the attached returned message.

                   The mail system

<hepler@vlsi-concepts.com>: host mailin-01.mgtinternet.com[204.90.71.225] said:
    553 5.0.0 <hepler@vlsi-concepts.com>... Message refused - see
    http://mail-cleaner.com/refused-1.html (in reply to RCPT TO command)
```

DrEdHepler, 2018-01-18:
Hi Olivier,

Thanks so much for your response!

bluetoothctl --version returns 5.44

I do see that gattlib_notification_start writes 0x0001 with the link 
given below, but when I download gattlib and perform the make, it uses 
the version of gattlib.c found in dbus.    Here, write_char_by_handle 
simply returns -1 and notification_start is significantly different!

To build, I created a "build" directory in gattlib-master, did a "cd" to 
it, then did "cmake ..".    I then did "make install".

To verify that it is using dbus/gattlib.c,  I renamed the file to 
gattlib.c.save, then did "make" again.   This time is fails since it 
can't find gattlib.c

This probably explains why I'm getting unusual results...    Did I 
create the "build" directory and do "cmake" in the wrong location?

Thanks again!

Ed Hepler

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

DrEdHepler, 2018-01-18:
Hi Olivier,

As an experiment, I created a "build" directory in gattlib-master/bluez, 
then did a "cd" to it and performed "cmake .." in this new location.

The cmake finished, but when I performed the "make", a number of errors 
were thrown.   The first error complains that "ATT_MAX_MTU" isn't 
defined.    It looks like this is defined in "att.h" for the Bluez4 
version, but not in the Bluez5 version...

Thanks again for your assistance...

Ed Hepler

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

oliviermartin, 2018-01-18:
No you were building well. I need to remember why I did not implement `gattlib_write_char_by_handle()` - but I think it was just not a priority. Can you not use `gattlib_write_char_by_uuid()` instead?

DrEdHepler, 2018-01-18:
I was using gattlib_write_char_by_handle() since I wanted to write to 
the "handle+1" to enable notifications.

But note that the new gattlib version also doesn't report handles when 
reporting characteristic information!

Is there a way to use gattlib_write_char_by_uuid() and specify a handle 
offset (so that the attribute info for notification enabling can be 
written directly)??

Thanks again for your assistance...

Ed

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

oliviermartin, 2018-01-18:
The functions `void gattlib_register_notification(gatt_connection_t* connection, gattlib_event_handler_t notification_handler, void* user_data);` + `int gattlib_notification_start(gatt_connection_t* connection, const uuid_t* uuid);` should do the job to register a handle and start the notification (by writing at the `handle+1`)

The new gattlib version does not report the handle number because the handle is hidden from the DBUS interface. I was initially thinking to use some virtual 'handle' for the DBUS interface but I did not do it (that is the reason I did not add support for `gattlib_write_char_by_handle()`)

DrEdHepler, 2018-01-18:
Hmm...    OK, those are the functions I have been using, but without luck.

I can use gatttool, to manually generate the proper sequence, and see 
the notifications arrive (through gatttool), but doing the same sequence 
(registering a notification handler, notification_start (then turning 
the service on via the write to the control characteristic), doesn't 
produce any notifications...

The operations behind the DBUS interface are hard to trace, so I'm 
having difficulty figuring out how to debug this...

Thanks for your assistance!

Ed Hepler

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

oliviermartin, 2018-01-18:
I guess you saw this example https://github.com/labapart/gattlib/blob/master/examples/notification/notification.c

DrEdHepler, 2018-01-18:
Olivier,

Looks like I stand corrected!

I modified the example "notification.c" to use my device, and that 
example works (produces notifications).

The only difference between the calls made in the example and my code is 
the inclusion of the GMainLoop stuff near the bottom of the example.

My code is embedded in a Tcl/Tk app and the calls to gattlib are made 
from there.   I didn't think I needed the GMainLoop stuff since Tcl/Tk 
provides the framework, but perhaps it is also intercepting the 
notifications...

Does gattlib depend on the GMainLoop framework in any way?

Thanks again!

Ed Hepler

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

oliviermartin, 2018-01-19:
Unfortunately, legacy Bluez and DBus depends on Glib (and so on GMainLoop). So your program must use `GMainLoop` or `g_main_context_iteration` (see: https://developer.gnome.org/programming-guidelines/stable/main-contexts.html.en).

I close this issue for now but fill free to continue the discussion in this thread or re-open this issue if you think it needs to.

closed 2018-01-19 by oliviermartin

DrEdHepler, 2018-01-19:
Olivier,

OK, thanks!!

I found a way to include calls to gtk_main_iteration() from within the 
Tcl main loop.    So, with your help, I've been able to get things 
working...

Thanks again,

Ed Hepler

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

DrEdHepler, 2018-01-24:
Hi Olivier,

Thanks again for your recent help.    Most things are now working for 
me, but I have a quick feature question.

When scanning for new devices, gattlib/bluez finds new devices and 
presents them via the API.   If a device is turned off, and a new scan 
is performed, the (now off) device still appears in the list.   I 
believe this is due to the device address/info being cached in bluez.

Is there a way (via gattlib) to flush the cache prior to running the 
scan so only currently advertising devices will be listed?

Thanks!

Ed

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com

nico202, 2018-02-18:
@DrEdHepler Have you found a way of clearing the cache?

DrEdHepler, 2018-02-18:
No...

It appears that bluez provides a way to do it as the tool "bt-device" 
has a "list" and "remove" option to help manage devices.

However, I haven't found a way to access this capability via gattlib...

-- 

--------------------------------------------------------------------------------------------------------------------------
VLSI Concepts offers synthesizable processor cores for embedded control 
applications.
See our standard products and ask about customization...
--------------------------------------------------------------------------------------------------------------------------
*Dr. Edward L. Hepler*
President, VLSI Concepts, Inc.               Adjunct Professor, 
Villanova University
     VLSI and System Architecture                Graduate Courses:
     Embedded System Design ECE-8440 System Design and Modeling
     C and Java Programming ECE-8445 Advanced Computer Architecture
     VHDL modeling and Simulation      ECE-8460 VLSI Design
email:   hepler@vlsi-concepts.com              or            
ehepler@villanova.edu
mobile: (484) 459-1126
www:   http://www.vlsi-concepts.com *Read: I Cor 8:6*

---
This email has been checked for viruses by AVG.
http://www.avg.com