labapart/gattlib #60: Returned an error when an address is malformed

oliviermartin, 2018-01-18:
I was informed by email that `gattlib` fails with the assertion:
`(process:1191): GLib-GIO-CRITICAL **: g_dbus_connection_signal_subscribe: assertion 'object_path == NULL || g_variant_is_object_path (object_path)' failed`
... when the address is malformed. In this case, there was a space before the address in the string.

Azkom, 2019-01-23:
When I use the comannd: ./discover fc:d6:bd:10:19:2a
**I have this failure:**
"Device connected error: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

Fail to connect to the bluetooth device."
When I use the command: ./discover FC:D6:BD:10:19:2A, there is no error and the connections is successful.

mamert, 2019-05-20:
For me, letter case does not seem to matter. I get the same error as @Azkom even when I run ./discover FC:D6:BD:10:19:2A  (yes, same MAC as he used).
./ble_scan lists my devices correctly.

Compiled and run on Raspbian Stretch

oliviermartin, 2019-05-20:
Hi @mamert, could you try again? I have made few changes/fixes today in `gattlib`.
I still believe the issue exists  but who knows (I saw a similar issue when trying to connect to many devices one after the other one).

mamert, 2019-05-21:
@oliviermartin I'm afraid I can only add to the confusion, and I apologize if I had missed something obvious... before pulling your changes today, I reset everything to defaults and tested again... and the old version worked. ./discover successfully connected to my device (the BT device now won't connect with my phone, but that's a different matter).
On the other hand, ./discover FC:D6:BD:10:19:2A (a MAC that doesn't exist here) still throws the same "method doesn't exist" error, even after using the updated gattlib. So, I guess it actually meant "device not found"? But the device doesn't need to be on, only paired (stored in /var/lib/bluetooth/{adaptermac}/) for there to be no error. If the device is on but not paired, we get the error.
Curiously, sudo l2ping gives "can't connect: Host is down" when ./discover works fine.