labapart/gattlib #65: value_handle stays '0'

willem4ever, 2018-03-02:
Scanning a device I get;

```
Succeeded to connect to the bluetooth device.
service[0] start_handle:00 end_handle:00 uuid:0x1801
service[1] start_handle:00 end_handle:00 uuid:49535343-fe7d-4ae5-8fa9-9fafd205e455
service[2] start_handle:00 end_handle:00 uuid:0x180a
characteristic[0] properties:20 value_handle:0000 uuid:0x2a05
characteristic[1] properties:18 value_handle:0000 uuid:49535343-4c8a-39b3-2f49-511cff073b7e
characteristic[2] properties:0c value_handle:0000 uuid:49535343-8841-43f4-a8d4-ecbe34729bb3
characteristic[3] properties:3c value_handle:0000 uuid:49535343-1e4d-4bd9-ba61-23c647249616
characteristic[4] properties:02 value_handle:0000 uuid:0x2a26
characteristic[5] properties:02 value_handle:0000 uuid:0x2a27
characteristic[6] properties:02 value_handle:0000 uuid:0x2a25
characteristic[7] properties:02 value_handle:0000 uuid:0x2a24
characteristic[8] properties:02 value_handle:0000 uuid:0x2a29
characteristic[9] properties:02 value_handle:0000 uuid:0x2a2a
characteristic[10] properties:02 value_handle:0000 uuid:0x2a23
characteristic[11] properties:02 value_handle:0000 uuid:0x2a28
```
i need a valid handle to use gattlib_write_char_by_handle ..... (e.g. when using nordic_uart example after modifying NUS_CHARACTERISTIC_TX_UUID) so this fails ....
however when I use gattlib_write_char_by_uuid  (used in read_write example) and pass the proper uuid I can write to the BLE device

BWhitten, 2018-11-23:
Without handle and attr_handle_start and attr_handle_end in services we cant make a sensible tree output with characteristics associated to services.

It also means in the case of duplicate characteristic UUID's in multiple services that you cant identify the correct one to write too,

XavierXin, 2019-03-19:
same issue

br101, 2019-05-28:
after i also found this issue in gattlib, i started to write my own library: https://github.com/infsoft-locaware/blzlib... 

it can differentiate UUIDs in different devices and is much faster. may it help you too, and i hope it's not considered impolite to post that here...

oliviermartin, 2019-06-12:
Value Handles are now supported in gattlib: https://github.com/labapart/gattlib/commit/2d81b253dae67388466cbb1f7534f56d959bd1cb

closed 2019-06-12 by oliviermartin