labapart/gattlib #66: Issue in CMake file

BikshapathiM, 2018-03-06:
Dear Owner,
In CmakeList.txt, it's taking the DBus api even though the bluez version is grater than 5.42.
check it once from line number 51. (gattlib/CmakeList.txt)
**actual Cmake code:**
if (BLUEZ_VERSION_MAJOR LESS 5)
  set(GATTLIB_DBUS FALSE)
elseif (BLUEZ_VERSION_MINOR LESS 42)
  if (GATTLIB_FORCE_DBUS)
    set(GATTLIB_DBUS TRUE)
  else()
    set(GATTLIB_DBUS FALSE)
  endif()
else()
set(GATTLIB_DBUS TRUE)
**Need to change :** 
if (BLUEZ_VERSION_MAJOR LESS 5)
  set(GATTLIB_DBUS TRUE)
elseif (BLUEZ_VERSION_MINOR LESS 42)
  if (GATTLIB_FORCE_DBUS)
    set(GATTLIB_DBUS TRUE)
  else()
    set(GATTLIB_DBUS FALSE)
  endif()
else()
set(GATTLIB_DBUS FALSE)

Please make this change in CMakeList.txt and update it properly.

oliviermartin, 2019-03-05:
Sorry @BikshapathiM, I am not sure to understand. Are you saying I should make this change?

```
if (BLUEZ_VERSION_MAJOR LESS 5)
set(GATTLIB_DBUS TRUE)
```

But in this case, it will take DBus API for Bluez v4.xx which is not supported yet.

oliviermartin, 2020-04-08:
Thanks @BikshapathiM for reporting this issue.
We now have travis-ci that builds the different versions of gattlib. And there is no compilation issue. I am assuming this issue is fixed. If it is not, feel free to open a new issue.

closed 2020-04-08 by oliviermartin