labapart/gattlib #75: How to get RSSI using gattlib

SamirMouhoune, 2018-11-18:
Hi, 

I just  started using gattlib and I would like to know if is there a way to get RSSI level using gattlib.

Thanks in advance for your support

dhavalhparikh, 2019-07-02:
Hi,
Any updates on this?
I would like to use this feature as well!

Thank you!

oliviermartin, 2019-07-09:
@SamirMouhoune and @dhavalhparikh  you might have seen I have just pushed a new function `gattlib_get_rssi()` to retrieve the RSSI. Let me know if you have any issue.

closed 2019-07-09 by oliviermartin

dhavalhparikh, 2019-07-09:
Hi @oliviermartin 
Thank you so much for your prompt reply and adding the new feature!
I was testing it and the **rssi always reads 0** for all devices that the `ble_scan.c` can connect to.

## Here's the code that I have:
```
 if (is_connected) {
        ret = gattlib_discover_char(gatt_connection, &characteristics, &characteristics_count);
        if (ret != 0) {
            gt_log("Fail to discover characteristics.\n");
            goto disconnect_exit;
        }

        // read rssi
	int16_t rssi = -1;
	if (GATTLIB_SUCCESS != gattlib_get_rssi(gatt_connection, &rssi)) {
		gt_log("\nERROR: could not read rssi!\n");
	}
	gt_log("rssi = %d\n", rssi);

        for (i = 0; i < characteristics_count; i++) {
            gattlib_uuid_to_string(&characteristics[i].uuid, uuid_str, sizeof(uuid_str));

            gt_log("characteristic[%d] properties:%02x value_handle:%04x uuid:%s\n", i,
                    characteristics[i].properties, characteristics[i].value_handle,
                    uuid_str);
        }
        free(characteristics);
    }
```

## Here's the output:
```
Scanning for BLE devices now:
Discovered 58:A0:D3:E4:40:40
Discovered 70:14:08:C5:53:03
Discovered 38:F9:D3:7C:93:5C
Discovered C6:97:77:10:49:34
Discovered 68:B6:67:0E:F6:94
Discovered 7B:76:9F:CD:01:96
Discovered 4A:72:5E:8B:6A:7E
Discovered 4E:BC:C8:F2:D9:5A
Discovered 6D:10:F4:05:53:16
Discovered C9:A5:D1:AC:27:2C
Discovered 59:D9:A7:C0:C9:5C
Discovered 59:9E:C8:73:56:45
Discovered F0:18:98:0C:21:2A - 'MacBook Pro'
Discovered 5F:1E:78:29:0F:4E
Discovered 7F:9C:41:88:55:78
Discovered 47:4F:12:02:53:14
Discovered 70:4A:73:D1:87:39
Discovered 6A:20:04:0C:3C:5F
Discovered 4C:BD:5D:CB:60:18
Discovered 79:9D:14:05:E8:EA
Discovered 65:C6:A8:53:69:16
Discovered 78:E2:9F:27:D4:9E
Discovered 7A:FD:BB:1E:E4:13
Discovered 5F:42:6E:80:37:1D
Scan completed
------------START 58:A0:D3:E4:40:40 ---------------
Device connected error (device:/org/bluez/hci0/dev_58_A0_D3_E4_40_40): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Fail to connect to the bluetooth device.
------------DONE 58:A0:D3:E4:40:40 ---------------
------------START 70:14:08:C5:53:03 ---------------
Device connected error (device:/org/bluez/hci0/dev_70_14_08_C5_53_03): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Fail to connect to the bluetooth device.
------------DONE 70:14:08:C5:53:03 ---------------
------------START 38:F9:D3:7C:93:5C ---------------
Device connected error (device:/org/bluez/hci0/dev_38_F9_D3_7C_93_5C): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Fail to connect to the bluetooth device.
[... 144 lines cut ...]
characteristic[4] properties:08 value_handle:0024 uuid:69d1d8f3-45e1-49a8-9821-9bbdfdaad9d9
characteristic[5] properties:18 value_handle:0010 uuid:af0badb1-5b99-43cd-917a-a77bc549e3cc
characteristic[6] properties:18 value_handle:000b uuid:8667556c-9a37-4c91-84ed-54ee27d90049
characteristic[7] properties:02 value_handle:001c uuid:0x2a0f
characteristic[8] properties:12 value_handle:0019 uuid:0x2a2b
characteristic[9] properties:20 value_handle:0007 uuid:0x2a05
characteristic[10] properties:18 value_handle:002e uuid:9b3c81d8-57b1-4a8a-b8df-0e56f7ca51c2
characteristic[11] properties:0a value_handle:0036 uuid:c6b2f38c-23ab-46d8-a6ab-a3a870bbd5d7
characteristic[12] properties:18 value_handle:0032 uuid:2f7cabce-808d-411f-9a0c-bb92ba96c102
------------DONE 78:E2:9F:27:D4:9E ---------------
------------START 7A:FD:BB:1E:E4:13 ---------------
Device connected error (device:/org/bluez/hci0/dev_7A_FD_BB_1E_E4_13): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Fail to connect to the bluetooth device.
------------DONE 7A:FD:BB:1E:E4:13 ---------------
------------START 5F:42:6E:80:37:1D ---------------
Device connected error (device:/org/bluez/hci0/dev_5F_42_6E_80_37_1D): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Fail to connect to the bluetooth device.
------------DONE 5F:42:6E:80:37:1D ---------------

```

Any idea why it's always reading 0?

Thank you,
Dhaval

oliviermartin, 2019-07-09:
Actually you are right, I have just tested `gattlib_get_rssi()` when the connection is established and it always returns `0`. This limitation needs to be investigated. I have disabled the API in the meantime: https://github.com/labapart/gattlib/commit/0142c33dd8fd1a094ea0a4d5ac19af86eee8cae7

I also introduced `gattlib_get_rssi_from_mac()` that can be used before the connection is established. This function returns the correct RSSI while the connection is not established.

BraydenBrekke, 2020-04-13:
@oliviermartin It seems as though gattlib_get_rssi_from_mac() is also always returning 0. 

```
int16_t  rssi;
void* adapter;
int ret;
int main() {
        ret = gattlib_get_rssi_from_mac(adapter,"DC:84:B2:0A:03:2C", &rssi);
        printf("%i\n", ret);
        printf("%i\n", rssi);
        return 0;
}
```

Output:
```
0
0
```
Is there something I am missing?

oliviermartin, 2020-04-14:
@BraydenBrekke Actually, you are right calling `gattlib_get_rssi_from_mac(adapter,"DC:84:B2:0A:03:2C", &rssi)` does not return anything - even on my computer. I would expect to see an error as the device `DC:84:B2:0A:03:2C` does not exist around me. It looks to me an issue in GDBUS library. GDBUS always returns a valid object even when the object path does not exist.
Unfortunately, I do not have enough personal time to investigate it further :-(

To return a valid RSSI for your device, you should at least have BLE scanning enable. Have a look at the function `gattlib_adapter_scan_enable()`.

BraydenBrekke, 2020-04-15:
@oliviermartin 

Thank you for the help! So I got the RSSI value, only after scanning. Does the gattlib_get_rssi_from_mac function require that I scan first?

```
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <gattlib.h>
#include <string.h>

int16_t  rssi;
void* adapter;

int ret;

 static void device_discover(void *adapter, const char* addr, const char* name, void *user_data){
    if (name){
     printf("%s\n", addr);
    }
 }

int main(){
    	ret = gattlib_adapter_open(NULL, &adapter);
        ret = gattlib_adapter_scan_enable(adapter, device_discover, 4,NULL);
    	ret = gattlib_adapter_scan_disable(adapter);
        printf("%i\n",ret);
        gattlib_get_rssi_from_mac(adapter,"DC:84:B2:0A:03:2C", &rssi);
        printf("%i\n", rssi);
    	ret = gattlib_adapter_close(adapter);

    return 0;
}
```
Output

```
DC:84:B2:0A:03:2C
0
-65
```

oliviermartin, 2020-04-15:
It is actually your BLE controller that requires to listen/scan devices or connect to a device to connect their/its RSSI. Without explicitely scanning, your BLE controller is off.

PR #113 (2019-07-09): gattlib: Introduced gattlib_get_rssi()
Fix: #75
commit eca8415f (2019-07-09): gattlib: Introduced gattlib_get_rssi()

Fix: #75