labapart/gattlib #77: Slow writes

BWhitten, 2018-12-18:
Hi,

I am using gattlib_write_char_by_uuid to write to a write no response characteristic, on a device running BlueZ 5.50 and I am seeing incredibly long delays between writes, 360-420ms typically, where as my Android mobile phone is able to achieve 0.7ms with the occasional 58ms between writes.
As seen on air with a BPA600 BLE sniffer.

I call gattlib_write_char_by_uuid with blocks of 20 bytes repeatidly, as thats is my max MTU.
I have tried calling it with the full block of data hoping that underlying will segment and send faster but the call errors out.

pdragoi, 2020-11-16:
Any updates on this? Having the same issue, writing (both with and without a response) takes too long.

alinouriz, 2021-06-02:
I have sort of the same problem. In Ubuntu 20, I try to write using "write without response" and in response turning LED connected to the Bluetooth device RN4871(microchip). All works well. The issue is if I do multiple writes at short intervals (below 500ms) my latency (from sensing the write command in my PC to LED turning ON) is below 20ms, which is fine for my application. However in longer intervals, lets say 1 second between two write command, the latency can go up to 300ms! I tried to figure this out by changing the connection protocols measuring the latency using different methods, all the same result!

I was wondering if anyone has any update on this issue.