labapart/gattlib #79: Multiple gatt peripheral connection at same time

andreaaizza, 2019-01-03:


andreaaizza, 2019-01-03:
Hi, 
any example code showing capability of connecting to multiple peripherals, sending notifications, at the same time? Does this library support it? 
Regards,
Andrea

dbonnell, 2019-10-28:
I have the same question.  Before I could get too far with testing, one of my BLE devices died and I've been sidetracked with trying to get gattlib working reliably with an emulated Nordic UART server on the nRF Connect app on Android.  (It connects, but very rarely does the notification on the TX char fire, even though I can see them being received in btmon).

If/when I figure out the multiple connections, I'll report back here.

dbonnell, 2019-10-29:
Ok, after fixing a bug in gattlib_connect with its timeout firing after it had finished and trashing things (or directly causing a segfault in g_main_loop_stop), notifications were working reliably for one device.  I then added a second device.

At first I got some weird results, like the callback being called twice (once per connection) when one device wrote.  I also got an error from bluez when trying to start notifications on the second connection that the characteristic was already registered.  I added some code in gattlib_notification_start to catch that error and ignore it, but then I just didn't get any notifications from the second device, as they hadn't been enabled.  (No write to the characteristic to enable them).   

Here's where it really got weird, it just started working.  Well, until the order of the device connections switched, then the first time it got the bluez "Already registered" error but then on the next attempt, both connections work perfectly.

I don't think I made any changes in either my code or gattlib after that, so no idea what the "magic" was.  Perhaps dbus is remembering something, not sure.

In any case, the answer is that multiple peripheral connections kind of work.