labapart/gattlib #80: write request without response not implemented

Haggui, 2019-01-04:
I need to implement an OTA using the gattlib librray, however to implement this functionalitie I need to send write request without response, looking into the gattlib_read_write I can see that it supports only write with response. I tried to add it and to recompile the hole library, however this disn't work ad I got an error message indicating "Fail to discover characteristics.". Could you please help me with this issue. Thank you

br101, 2019-05-28:
maybe try this alternative: https://github.com/infsoft-locaware/blzlib... i supports write without response...

oliviermartin, 2019-07-05:
@Haggui  I added Write-Without-Response in this commit: https://github.com/labapart/gattlib/commit/8020503ab697837ba4cebc58e3441183ee1f095d

I also demonstrate the support by connecting to Nordic Semiconductor's board running UART sample: https://github.com/labapart/gattlib/blob/master/examples/nordic_uart/nordic_uart.c#L129

The change in your application code is straightforward.

closed 2019-07-05 by oliviermartin