labapart/gattlib #93: Segmentation fault when reading a characteristic after reconnection

yghdj, 2019-03-27:
Hi, and thank you for your work.

I'm working on implementing a Linux daemon that reads plain text on a socket and writes it to a BLE device. I'm using GattLib with BlueZ version 5.48 and DBus interface on Ubuntu 18.04. The BLE device can respond to the daemon by notifying it on another characteristic. So far, it's been working quite good, but I'm facing some problems upon reconnection. 

When I connect to the device for the first time I've got no problem. I can write on the characteristic without any problem. But if I disconnect using `gattlib_disconnect()`, and then reconnect using `gattlib_connect()`, then receiving notifications from the device will result on a segmentation fault, always. Using valgrind and gdb, I've found that the segfault occurs in function `on_handle_characteristic_property_change()`, and more precisely, when it tries to read the `gatt_connection_t` pointer provided to this function as the `user_data` argument. 

Valgrind tells me the object pointed has been free'd during `gattlib_disconnect()` so I suppose the argument provided to `on_handle_characteristic_property_change` is the pointer to the previous connection. But why would it be? I used `gattlib_notification_start()` with the new connection's `gatt_connection_t` object before. Is the previous connection pointer still stored somewhere?

Hope you can help me

Thank you

johndoe31415, 2019-09-28:
Hey there! I've been having similar issues with reconnection, reported in https://github.com/labapart/gattlib/issues/132 -- do you have any pointer on how you solved your problem back then? It seems like this could be the same root cause.

dbonnell, 2019-10-28:
gattlib_notification_start registers a signal handler for g-properties-changed on the characteristic and passes the connection as user_data.  That handler is never cleaned up by gattlib_disconnect (or gattlib_stop_notification) and the next time that handler is invoked, on_handle_characteristic_property_change will very likely get a NRE when it attempts to de-reference connection->notification_handler, as connection has been free'd (by gattlib_disconnect).

Each handler connected in gattlib_notification_start needs to be saved on the connection (e.g. connection->notification_handlers, an array of {uuid_t*, gulong}).   Then gattlib_stop_notification can disconnect one and gattlib_disconnect can disconnect them all, before destroying the connection and its metadata.  (Must store a copy of the provided uuid_t*, as the original may go out-of-scope before it is needed by gattlib_stop_notification or gattlib_disconnect).

oliviermartin, 2020-04-08:
Thanks @yghdj for reporting this issue. It should now be fixed by this PR (thanks to @johndoe31415 's test file that really helped me to fix this issue): https://github.com/labapart/gattlib/pull/163

closed 2020-04-08 by oliviermartin