labapart/gattlib #104: arm-openwrt-linux-muslgnueabi/bin/ld: cannot find -ltinfo

rt005744, 2019-05-27:
arm-openwrt-linux-muslgnueabi/bin/ld: cannot find -ltinfo

rt005744, 2019-05-27:
Execute the cmake command to return an error，How to solve this problem？

oliviermartin, 2019-05-27:
@rt005744 Could you provide steps to duplicate this issue?

rt005744, 2019-05-28:
root@TinaLinux:/mnt/UDISK/gattlib/build# ./examples/read_write/read_write DC:2C:
26:01:E2:27 write fb349b5f-8000-0080-0010-0000f1ff0000 0x1234
Value to write: 0x1234
Fail to find handle for UUID.
Could not find GATT Characteristic with UUID 0000fff1-0000-1000-8000-00805f9b34fb

 why return Could not find GATT Characteristic with UUID 0000fff1-0000-1000-8000-00805f9b34fb?
I write to fb349b5f-8000-0080-0010-0000f1ff0000, but return 0000fff1-0000-1000-8000-00805f9b34fb, why?

oliviermartin, 2019-07-05:
I did some investigation and it looks like your issue is related to Big-Endian support. Probably the main file to change for adding Big-Endian is https://github.com/labapart/gattlib/blob/master/dbus/bluez5/lib/uuid.c
I do not have Big-Endian hardware with me.

If you want me to add Big-Endian please contact me through https://labapart.com/about/