labapart/gattlib #111: Connection fails with Authentication Failure

cryptax, 2019-07-08:
I am trying to connect to a device using **sec-level high** and **random** address type. 
This fails and hcidump shows reason "Authentication Failure".

```c
unsigned long options = GATTLIB_CONNECTION_OPTIONS_LEGACY_BT_SEC_HIGH | GATTLIB_CONNECTION_OPTIONS_LEGACY_BDADDR_LE_RANDOM;
char device_address[] = "D2:A7:4C:76:F3:E0";

return gattlib_connect(NULL, &device_address[0], options);
```

This is the hcidump:

```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr D2:A7:4C:76:F3:E0 type 1
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 24 max_interval 40
    latency 0 supervision_to 42 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr D2:A7:4C:76:F3:E0 (Random)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 14
    LE Read Remote Used Features (0x08|0x0016) ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28
> HCI Event: Command Status (0x0f) plen 4
    LE Start Encryption (0x08|0x0019) status 0x00 ncmd 1
> HCI Event: Encrypt Change (0x08) plen 4
    status 0x08 handle 64 encrypt 0x00
    Error: Connection Timeout
< HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x05
    Reason: Authentication Failure
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x12 ncmd 1
    Error: Invalid HCI Command Parameters
```

I am on a **RPI3** with Raspbian 9.9 and **bluez 5.43**.

oliviermartin, 2019-07-08:
Could you try to connect to it using `bluetoothctl`? If it works, can you send me the `bluetoothctl` logs?
Does it work from another computer?

I suspect this issue to be specific to the BLE adapter/kernel/bluez and has nothing to do with `gattlib`.

cryptax, 2019-07-09:
To summarize quickly, **it manages to connect** with `bluetoothctl` or with another host and gattlib, but the **connection apparently does not use the high security level** which is mandatory in my case. 

>Could you try to connect to it using bluetoothctl?

Yes, with bluetoothctl it works, though I haven't specified "high" security level (don't know how to). Instead, I paired the device. But I see I get an "Insufficient authentication" while trying to access a characteristic, which means I am not in the right security level.

```
[bluetooth]# connect D2:A7:4C:76:F3:E0
Attempting to connect to D2:A7:4C:76:F3:E0
[CHG] Device D2:A7:4C:76:F3:E0 Connected: yes
Connection successful
[NEW] Primary Service0]# 
	/org/bluez/hci0/dev_D2_A7_4C_76_F3_E0/service0008
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[D2A74C76F3E0]# pair D2:A7:4C:76:F3:E0
Attempting to pair with D2:A7:4C:76:F3:E0
[CHG] Device D2:A7:4C:76:F3:E0 Paired: yes
Pairing successful
[D2A74C76F3E0]#  select-attribute /org/bluez/hci0/dev_D2_A7_4C_76_F3_E0/service000c/char0013
[D2A74C76F3E0:/service000c/char0013]# write 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF
Attempting to write /org/bluez/hci0/dev_D2_A7_4C_76_F3_E0/service000c/char0013
```

This is what hcidump sees:

```
> ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Write req (0x12) on handle 0x0014
```

>Does it work from another computer?

On another computer (Linux Mint 18.3), I installed your gattlib, compiled my sample program, and I manage to connect, but same I don't think it does it right either because then afterwards, when I try to write a given characteristic, I get "Error: Insufficient authentication (5)" which should not occur as I set `GATTLIB_CONNECTION_OPTIONS_LEGACY_BT_SEC_HIGH`... so maybe it has connected, but not properly would be my guess :(

```
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x00 (passive)
    interval 60.000ms window 30.000ms
    own address: 0x00 (Public) policy: white list only
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr D2:A7:4C:76:F3:E0 (Random)
      Flags: 0x06
      Unknown type 0x06 with 16 bytes data
      Unknown type 0xff with 8 bytes data
      RSSI: -71
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr D2:A7:4C:76:F3:E0 type 1
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 400 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 69, role master
      bdaddr D2:A7:4C:76:F3:E0 (Random)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 69
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
< ACL data: handle 69 flags 0x00 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0xffff
      type-uuid 0x2803
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
...
< ACL data: handle 69 flags 0x00 dlen 15
    ATT: Write req (0x12)
      handle 0x0014 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
> ACL data: handle 69 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Write req (0x12) on handle 0x0014
```

oliviermartin, 2019-07-09:
For Bluez >= v5.42, I use the Bluez DBUS API. And there is no way from this API to specify the security level: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt

What you might need to do is to pair the device before using `gattlib`. You can pair your device either using the GUI of your OS or using `bluetoothctl` with the command `pair`.
If it works, I might consider adding the pairing API to gattlib.

cryptax, 2019-07-09:
Everything is horribly buggy. I don't get the same answer with the same tools every day :( That being said, when I connect and pair (with `bluetoothctl`), it always works "better", i.e I connect with gattlib (at least), but write fails.

```
[+] Connected
Error while writing GATT characteristic (rc: 2)
```

On another host - same program - I don't get the same error. It looks on sending the write packets and receiving "Insufficient authentication" ... I guess you are right that the problem is not with gattlib but with the underlying bluez or dongle. I am contacting the bluetooth list. Thanks.

oliviermartin, 2019-07-15:
I assume this issue now fixed: https://marc.info/?l=linux-bluetooth&m=156294282207207 :smile:

closed 2019-07-15 by oliviermartin