labapart/gattlib #115: can scan (permission issue)

zewenhs, 2019-07-10:
It come error when I execute `ble_scan` command(with sudo)
` sudo ./examples/ble_scan/ble_scan`

> ERROR: Set scan parameters failed (are you root?).
> ERROR: Failed to scan.

How to avoid this error? thanks!

oliviermartin, 2019-07-10:
Which version of Bluez are you using (`bluetoothctl --version`)? Recent version of Bluez do not require to launch `ble_scan` as root?

zewenhs, 2019-07-11:
I have tried ubuntu 14.04 with bluez 5.44 and ubuntu 16.04 with 5.48, all have this NG result!

oliviermartin, 2019-07-11:
How do you build `gattlib`? Can you also send me the build logs?
Do you confirm you have a bluetooth controller plugged on your PC and also the bluetooth adapter is enabled/power on?

zewenhs, 2019-07-12:
I build `gattlib` follow the method mentioned in `README` file, the following command and log are on raspi 3B:
[image]

**Also I have checked `bluetoothctl` command:**

[image]

**cmake log:**
[cmake.log](https://github.com/labapart/gattlib/files/3384609/cmake.log)

**build log**
[build.log](https://github.com/labapart/gattlib/files/3384610/build.log)

I hvae a question, the bluez vesion is 5.50 in my raspi but gattlib say it's 5.23(show in the first picture which say `build gattlib for bluez v5.23`)?
[image]

oliviermartin, 2019-07-12:
How did you install bluez v5.50? Which Linux distribution are you using?

zewenhs, 2019-07-13:
I install the official operating system which name is `Raspbian`, baseed on `debian`, and can be find [**here**](https://www.raspberrypi.org/downloads/)
[image]

And I follow this guide to install bluez 5.50: [Deploying BlueZ v5.50 on
Raspberry Pi 3 – Update](https://www.bluetooth.com/blog/bluez-on-raspberry-pi-update/)

oliviermartin, 2019-07-14:
The build system gets confused between the two versions of Bluez installed on your system.

I have just added support to force building `gattlib` for a specific version. In your case:

```
rm -Rf *
cmake -DBLUEZ_VERSION=5.50 .. && make
```

oliviermartin, 2019-07-15:
I will assume this new build option should fix your issue. Feel free to comment it or re-open this issue if you have still got an issue.

closed 2019-07-15 by oliviermartin