labapart/gattlib #121: dbus_connection_set_watch_functions

zewenhs, 2019-07-17:
First of all, I'm sorry for post this question, because it's not a bug about `gattlib`, but a problem I've encountered when I read bluez source code.
I am reading the `bluetoothctl` source code which is included in bluez, and want to use gdbus to control ble by myself. when i read this:
```
dbus_connection_set_watch_functions(conn, add_watch, remove_watch,
						watch_toggled, conn, NULL);

dbus_connection_set_timeout_functions(conn, add_timeout, remove_timeout,
						timeout_toggled, NULL, NULL);

dbus_connection_set_dispatch_status_function(conn, dispatch_status,
								NULL, NULL);
```
My question is when these callbacks(`add_watch`,`remove_watch`, `watch_toggled`, `add_timeout`,`remove_timeout`, `timeout_toggled`, `dispatch_status` ) will be invoked?

I have doing some verification: I add some print logs in these functions, and type `bluetoothctl` command and only press enter key, I found `add_watch` and `remove_watch` have been invoked for several times. In the DBus document there have following info:
```
dbus_connection_set_watch_functions():

Sets the watch functions for the connection.
These functions are responsible for making the application's main loop aware of file descriptors that need to be monitored for events,
```

`Sets the watch functions for the connectio`:In my understanding, this connection is not changed , why `add_watch` `remove_watch` have been invoked for several times? so what's the exactly this 'watch' function watch for(also 'timeout')? and Under what conditions will these functions be called? thanks

oliviermartin, 2019-07-17:
Any reason to not use gattlib as a frontend for Bluez?

zewenhs, 2019-07-28:
Gattlib is amazing and helpful, I want to know the details of the code implementation. 
the difference between knowing the name of something and knowing something.

oliviermartin, 2019-07-28:
No problem to help you with your specific problem through consultancy. My own company provide consultancy and services: https://labapart.com/services/

Feel free to contact us: https://labapart.com/about/

closed 2019-07-28 by oliviermartin