labapart/gattlib #125: cmake ble_scan. None of the required gattlib found

djok0127, 2019-07-22:
cmake 
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-gcc
-- Check for working C compiler: /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-g++
-- Check for working CXX compiler: /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/bin/arm-dey-linux-gnueabi/arm-dey-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'gattlib'
CMake Error at /opt/dey/2.4-r1/sysroots/x86_64-deysdk-linux/usr/share/cmake-3.8/Modules/FindPkgConfig.cmake:641 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  CMakeLists.txt:26 (pkg_search_module)

-- Configuring incomplete, errors occurred!
See also "/home/insight1/Git_Repositories/Insight.Core.ArcAgent.SBC/Firmware.BluetoothInvestigation/gattlib/examples/ble_scan/CMakeFiles/CMakeOutput.log".

oliviermartin, 2019-07-28:
You have not installed `gattlib` as a package on your computer. Either download [the latest package](https://github.com/labapart/gattlib/releases/tag/dev) or [build the package yourself](https://github.com/labapart/gattlib#package-gattlib)

closed 2019-07-28 by oliviermartin

Woody00h, 2020-03-14:
How do u solve the issue?
I install the gattlib_dbus_0.2-dev_x86_64.deb. When I build the ble_scan, got the same issue.

lucy0828, 2021-03-18:
I have the same issue when building notification.c 
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:741 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'glib-2.0'
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:741 (message):
  None of the required 'glib-2.0' found
Call Stack (most recent call first):
  CMakeLists.txt:28 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
-- Configuring incomplete, errors occurred!
See also "/home/pi/Thingy52_BLE/gattlib_0.2-dev_x86_64/build/CMakeFiles/CMakeOutput.log".