labapart/gattlib #126: Python Dependency

djok0127, 2019-07-30:
Hello, 
I am trying to implement gattlib into an SBC(using Linux Kernel).
I am aware of all the dependencies and have all of the figured out.
One of the dependency, i.e. Python3 is also installed, but I guess gattlib requires libpython3.7m.so.1.0.

I have Python 3.5 but I guess that does not fulfil the dependency for Python3.
Do you have any insight on how to get around this?

Note: I tried to find libpython3.7m.so.1.0, but I could not find 32bit one for the linux distribution.

Thank you for your time!

oliviermartin, 2019-08-05:
`gattlib` does not depend on Python 3.7. It uses python 3 if installed on your system. Could you send me the result of `find /usr/ -name 'python*.pc'`?
I am guessing you have a file named `python-3.7m.pc` that said you have a file `libpython3.7m.so.1.0`.

Which distribution and CPU architecture are you using?

joonazan, 2020-02-03:
Why does it use Python? I thought this was a pure C library.

makosolo, 2021-05-07:
root@solo-PC:/home/workspace/gattlib/build# cmake -DGATTLIB_FORCE_DBUS=TRUE ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen missing components:  dot
-- Checking for one of the modules 'bluez'
-- Checking for one of the modules 'glib-2.0'
-- Checking for one of the modules 'gio-unix-2.0'
Build DBus gattlib for Bluez v5.50
-- Could NOT find Python (missing: Python_LIBRARY Python_INCLUDE_DIR Development) (found version "2.7.16")
Install Sphinx to generate Python documentation
-- Checking for one of the modules 'gattlib'
-- Checking for one of the modules 'libpcre'
Package Gattlib for development version 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/workspace/gattlib/build

Be troubled by dependence！

makosolo, 2021-05-12:
I installed these dependencies and can compile through:
apt install cmake build-essential scons
apt install pkg-config
apt install libbluetooth-dev libreadline-dev
apt install glib-2.0 libglib2.0-dev
apt install doxygen
apt-get install libgtk2.0-dev
apt-get install uuid-dev