labapart/gattlib #129: doxygen missing components:  dot

rt005744, 2019-08-07:
ny@Tina:~/gattlib/build$ cmake ..
-- The C compiler identification is GNU 6.4.1
-- The CXX compiler identification is GNU 6.4.1
-- Check for working C compiler: /home/ny/tina/prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain/bin/arm-openwrt-linux-gcc
-- Check for working C compiler: /home/ny/tina/prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain/bin/arm-openwrt-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/ny/tina/prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain/bin/arm-openwrt-linux-g++
-- Check for working CXX compiler: /home/ny/tina/prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain/bin/arm-openwrt-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- Checking for one of the modules 'bluez'
CMake Warning (dev) at CMakeLists.txt:59 (set):
  implicitly converting 'BOOLEAN' to 'STRING' type.
This warning is for project developers.  Use -Wno-dev to suppress it.

Build gattlib for Bluez v4.101
-- Checking for one of the modules 'glib-2.0'
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.8.16 (fd73bd3a845fc1d9b4678eabebbd4457f376928e)") found components:  doxygen missing components:  dot
-- Checking for one of the modules 'gattlib'
-- Checking for one of the modules 'libpcre'
Package Gattlib for development version 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ny/gattlib/build

rt005744, 2019-08-07:
why “doxygen missing components:  dot”

oliviermartin, 2019-08-07:
It looks to be a Doxygen specific message. Anyway, we do not need `dot` and we can build without it. If you type `make` after your `cmake` it should build.

I had a quick look if I could remove it but I did not manage.

closed 2019-08-07 by oliviermartin

jakesyl, 2020-01-14:
Anyone find out what dot is?

Jedzia, 2020-04-05:
> 
> 
> Anyone find out what dot is?

"dot" is from Graphviz of https://www.graphviz.org/
Graphviz is open source graph visualization software and doxygen uses it to produce its inheritance diagrams, etc.

Often 
    
    find_package(Doxygen REQUIRED dot)
is used in CMake and fails to configure without Graphviz installed. 

    find_package(Doxygen REQUIRED)
still produces the “doxygen missing components: dot” warning message without failing to configure.

Either way:
- install from https://www.graphviz.org/download/ , or
- pacman -S mingw64/mingw-w64-x86_64-graphviz, or
- apt install graphviz, or
- etc.

guanshaoheng, 2020-09-27:
After `sudo apt install graphviz` the error comes to `Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen dot `