labapart/gattlib #132: Old connection data leaks over to new connections

johndoe31415, 2019-09-28:
Hi there,

I'm using gattlib 55d875438b9537c7536f7f0a9cdf5584b963d8f5 on a Raspberry Pi 3B+. I've modified the notification application so that it continuously tries to connect to a heart rate monitor and then from there on handle the callback data appropriately.

I find that once a connection is lost and a second one re-established, that the notification handler is called twice. If the connection is lost a second time and re-established, it is called three times for each notification. I.e., I'm assuming that the registration of notification handlers is cumulative, even though I've properly disconnected the libgatt connection altogether. Coincidentially in my case, the new `gatt_connection_t*` instances land at the same memory address as the first one; not sure if gattlib keeps a library-internal memory buffer for that or if it's just pure chance (in which case it might likely crash if I expanded/changed the program a bit).

It is my understanding that gattlib_register_notification is bound to a specific connection context (`gatt_connection_t*`) and that once it's disconnected using `gattlib_disconnect` it's not called anymore.

I've attached a small example application that demonstrates the issue. It's not pretty at all, but it was just a quick & dirty test of everything: [not.c](https://github.com/labapart/gattlib/files/3665464/not.c.txt)

Note the output I receive:

```
Attempting connection
Device connected error (device:/org/bluez/hci0/dev_20_C3_8F_FF_0A_7E): GDBus.Error:org.bluez.Error.Failed: Operation already in progress
Failed to connect to the bluetooth device.
Attempting connection
Connected: 0x75a005d0
Still connected...
Still connected...
Still connected...
Notification Handler 0x75a005d0: 04 00 
Still connected...
Notification Handler 0x75a005d0: 04 00 
Still connected...
[...]
Notification Handler 0x75a005d0: 04 00 
Still connected...
Still connected...
Still connected...
Still connected...
Still connected...
Still connected...
Disconnected 0x75a005d0
Finished everything, retrying...
Attempting connection
Connected: 0x75a005d0
Still connected...
Still connected...
Notification Handler 0x75a005d0: 06 00 
Notification Handler 0x75a005d0: 06 00 
Still connected...
Still connected...
Notification Handler 0x75a005d0: 06 00 
Notification Handler 0x75a005d0: 06 00 
Still connected...
```

Please advise if I'm doing something wrong or if this is indeed a bug.

Thank you,
Johannes

johndoe31415, 2019-09-28:
Interesting info: Something is definitly wrong there. I've compiled everything with libasan to check for reasons and this caused the second "connect" call to return a different pointer than the first one. Now I receive:

```
Still connected...
Notification Handler 0x74101ed0: 04 00 
Still connected...
Notification Handler 0x74101ed0: 04 00 
Still connected...
Still connected...
Still connected...
Still connected...
Still connected...
Still connected...
Disconnected 0x74101ed0
[...]
Attempting connection
Connected: 0x741179d0
Still connected...
Notification Handler 0x74101ed0: 06 00 
Notification Handler 0x741179d0: 06 00 
Still connected...
Notification Handler 0x74101ed0: 06 00 
Notification Handler 0x741179d0: 06 00 
Still connected...
```

I.e., the old connection (0x...ed0) is still dangling somehow. Odd.

dumanovskis, 2019-11-04:
I've had the same problem, a reconnect will cause notifications to trigger twice, a second reconnect would make them trigger three times, etc. This is regardless of whether subsequent `gattlib_connect` calls return the same pointer as the first one or not.

The issue seems to stem from the way gattlib registers notifications. A notification connects `on_handle_characteristic_property_change` to the specific GATT characteristic.

```
g_signal_connect(dbus_characteristic.gatt,
 "g-properties-changed",                                                                          
 G_CALLBACK (on_handle_characteristic_property_change),                                           
 connection);
```
The signal can only get disconnected from `gattlib_notification_stop`, which doesn't get called when disconnecting, no matter if it's spontaneous or with `gattlib_disconnect`.

For some reason adding a call to `gattlib_notification_stop` in the disconnection callback still doesn't help, but I haven't investigated that in detail.

Seems like gattlib needs to make gattlib_connection_t track any notifications that have been installed, and clean them up correspondingly both for `gattlib_disconnect` and spontaneous disconnect cases/

johndoe31415, 2019-11-04:
This confirms the suspicion that this is very likely a bug in gattlib.

I've worked around this in my application by jailing gattlib into its own address space using a subprocessing, then communicating its findings back through a pipe to the parent. If anything goes wrong and gattlib chokes, it gets forefully terminated and restarted. It's trapped in its containment at all times.

Example is here: https://github.com/johndoe31415/btlehrservice

dumanovskis, 2019-11-04:
My application only connects to one device, so a simple workaround is to only start the notification once, and then skip adding it if a reconnect is needed.

A solution in gattlib itself might be along the lines of adding a list to gattlib_context_t, and adding items to that list for each `g_signal_connect` call so that they may be used to perform `g_signal_handler_disconnect` when the connection is closed/lost.

oliviermartin, 2019-11-05:
I will try to find time to have a look in the next few days. Thanks for reporting the issue.

oliviermartin, 2020-04-08:
Thanks a lot @johndoe31415 for your source file. It really helped me to fix this issue.

Here is the PR that fixes your issue: https://github.com/labapart/gattlib/pull/163

Thanks again @johndoe31415 and @dumanovskis for your help to find this issue.

closed 2020-04-08 by oliviermartin