labapart/gattlib #138: Segmentation fault using gattlib_adapter_scan_enable_with_filter ()

dhavalhparikh, 2019-11-19:
Hi @oliviermartin 
Hope you are doing well!
I'm trying to use the scan filter and wanted to filter devices based on the RSSI and I'm running it on Raspberry Pi. 
The example works fine as is and I'm able to scan the devices but when I try to filter using the RSSI, I get a seg_fault.

In your `ble_scan.c` example,
I replaced this:
`ret = gattlib_adapter_scan_enable(adapter, ble_discovered_device, BLE_SCAN_TIMEOUT, NULL);`
with this:
`ret = gattlib_adapter_scan_enable_with_filter(adapter, NULL, -30, GATTLIB_DISCOVER_FILTER_USE_RSSI, ble_discovered_device, BLE_SCAN_TIMEOUT, NULL);`

Basically I'm trying to scan for devices and filter with `RSSI threshold of -30 dB`

However I get the following `seg fault:`
```
(process:9556): GLib-CRITICAL **: g_variant_ref: assertion 'value->ref_count > 0' failed
(process:9556): GLib-CRITICAL **: g_variant_get_type_string: assertion 'value != NULL' failed
Segmentation fault
```

I see that `gattlib_adapter_scan_enable()` is a wrapper around `gattlib_adapter_scan_enable_with_filter()` with 0 rssi threshold and `GATTLIB_DISCOVER_FILTER_USE_NONE`.

Any ideas what might be going on here?
Looking forward to hearing from you soon!

Thank you,
Dhaval

dumanovskis, 2019-11-20:
This is due to the 

```
g_variant_unref(rssi_variant);
```
call in gattlib_adapter.c. The `rssi_variant` gets freed before it actually gets used in the DBus call, so the filtering is likely to segfault.

dhavalhparikh, 2019-11-20:
@dumanovskis Thanks for getting back to me on this issue!
Do you know how we can prevent that?

dumanovskis, 2019-11-20:
The Gnome documentation site is currently down, and I am not immediately sure when to properly unreference elements that go into a GVariantBuilder.

You can fix the crash by changing the relevant_part in gattlib_adapter.c like this:

```
         GVariant *rssi_variant = NULL;
         if (enabled_filters & GATTLIB_DISCOVER_FILTER_USE_RSSI) {
                 GVariant *rssi_variant = g_variant_new_int16(rssi_threshold);
                 g_variant_builder_add(&arg_properties_builder, "{sv}", "RSSI", rssi_variant);
         }       

         org_bluez_adapter1_call_set_discovery_filter_sync((OrgBluezAdapter1*)adapter,
                         g_variant_builder_end(&arg_properties_builder), NULL, &error);
         if (rssi_variant) {
             g_variant_unref(rssi_variant);
         }
```

I'm not sure if that's fully correct from a glib point of view though.

dhavalhparikh, 2019-11-20:
I tried your solution. It works fine now and it filters the scan but I'm still getting this message:
`(process:1667): GLib-CRITICAL **: g_variant_unref: assertion 'value->ref_count > 0' failed`

dumanovskis, 2019-11-21:
Yes, it means some object is having `g_variant_unref` called that doesn't need to. Since I'm not sure when to properly unref a variant inside a builder, I'll leave the real solution here to Olivier or other devs who know what they're doing :)

oliviermartin, 2020-04-08:
I fixed the RSSI such as suggest @dumanovskis in this [commit](https://github.com/labapart/gattlib/commit/52f195f5c2cb98aee8dca02bd1b29b977d64b886) - his fix looks good to me. But I could not duplicate your other issue @dhavalhparikh .

Would you mind to run your test/app with GDB and send me the trace? To get the trace:
1. Start your app with GDB and GLIB option "warning as fatal" with `G_DEBUG=fatal-warnings gdb ./examples/ble_scan/ble_scan`
2. In gdb, type `run` to start your app
3. When GDB stop because of the error, type `where`. You should see a trace similar to this one:

```
Thread 1 "find_eddystone" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) where
#0  0x00007ffff74cfe97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff74d1801 in __GI_abort () at abort.c:79
#2  0x00007ffff78f5315 in g_assertion_message () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff78f537a in g_assertion_message_expr () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff791167e in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff791174b in g_variant_type_info_get_type_string () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff71b792e in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#7  0x00007ffff71b7a08 in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#8  0x00007ffff71b74a5 in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#9  0x00007ffff71ba150 in g_dbus_message_to_blob () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#10 0x00007ffff71ae63c in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#11 0x00007ffff71afaa6 in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#12 0x00007ffff71b2b12 in g_dbus_connection_send_message_with_reply () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#13 0x00007ffff71b33cb in g_dbus_connection_send_message_with_reply_sync () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#14 0x00007ffff71b37cf in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#15 0x00007ffff71bfa03 in  () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#16 0x00007ffff71c0dd4 in g_dbus_proxy_call_sync () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#17 0x00007ffff7badf61 in org_bluez_adapter1_call_set_discovery_filter_sync (proxy=0x55555576e0a0, arg_properties=0x7fffe4007ec0, cancellable=0x0, error=0x7fffffffd9a0)
    at /home/olivier/labapart/dev/gattlib/build/dbus/org-bluez-adaptater1.c:1450
#18 0x00007ffff7ba99db in gattlib_adapter_scan_enable_with_filter (adapter=0x555555785790, uuid_list=0x7fffffffdba0, rssi_threshold=-30, enabled_filters=3, discovered_device_cb=0x7ffff7bac724 <on_eddystone_discovered_device>, timeout=20, user_data=0x7fffffffdb90) at /home/olivier/labapart/dev/gattlib/dbus/gattlib_adapter.c:186
#19 0x00007ffff7bac8cd in gattlib_adapter_scan_eddystone (adapter=0x555555785790, rssi_threshold=-30, eddystone_types=18, discovered_device_cb=0x555555554d4a <on_eddystone_found>, timeout=20, user_data=0x0)
    at /home/olivier/labapart/dev/gattlib/common/gattlib_eddystone.c:69
#20 0x0000555555554ff0 in main (argc=1, argv=0x7fffffffdd48) at /home/olivier/labapart/dev/gattlib/examples/find_eddystone/find_eddystone.c:101
```

Please re-open a new issue with the new bug and I will be happy to have a look.

closed 2020-04-08 by oliviermartin