labapart/gattlib #140: unable to unregister dbus signal for ble notifications

rial01, 2019-12-27:
`
int gattlib_notification_start(gatt_connection_t* connection, const uuid_t* uuid) {
    ....
    // Register a handle for notification
	g_signal_connect(dbus_characteristic.gatt,
		"g-properties-changed",
		G_CALLBACK (on_handle_characteristic_property_change),
		connection);
    ......
}

int gattlib_notification_stop(gatt_connection_t* connection, const uuid_t* uuid) {
    ......
    g_signal_handlers_disconnect_by_func(
			dbus_characteristic.gatt,
			G_CALLBACK (on_handle_characteristic_property_change),
			connection);
    ......
}
`
In these two functions, these two addresses for dbus_characteristic.gatt are not the same!
In fact, we can not to unregister dbus signal for ble notifications.

oliviermartin, 2020-04-07:
I tried to duplicate your issue and I cannot do it. I was hoping you found a bug but it does not seem to be the case - or at least I cannot duplicate it.
I confirm both `dbus_characteristic.gatt` addresses are different, but it does not seem to bother `g_signal_handlers_disconnect_by_func()`.

If you provide a source file with a test, I could try to duplicate the issue again. But again, with the test I have done to validate this issue, I cannot duplicate the issue.

Here is the test:

```
	gatt_connection_t* gatt_connection;
	gattlib_characteristic_t* characteristics;

	gatt_connection = gattlib_connect(NULL, g_addr, GATTLIB_CONNECTION_OPTIONS_LEGACY_BDADDR_LE_RANDOM);
	if (gatt_connection == NULL) {
		fprintf(stderr, "Fail to connect to the bluetooth device.\n");
		goto EXIT;
	}

	ret = gattlib_discover_char(gatt_connection, &characteristics, &characteristics_count);
	if (ret != 0) {
		fprintf(stderr, "Fail to discover characteristics.\n");
		goto EXIT;
	}
	for (size_t i = 0; i < characteristics_count; i++) {
		gattlib_uuid_to_string(&characteristics[i].uuid, uuid_str, sizeof(uuid_str));

		printf("characteristic[%zu] properties:%02x value_handle:%04x uuid:%s\n", i,
				characteristics[i].properties, characteristics[i].value_handle,
				uuid_str);
	}
	free(characteristics);

	uuid_t temp_uuid;
	const char temp_uuid_str[] = "ef680201-9b35-4933-9b10-52ffa9740042";

	ret = gattlib_string_to_uuid(temp_uuid_str, strlen(temp_uuid_str) + 1, &temp_uuid);
	if (ret != 0) {
		fprintf(stderr, "Fail to convert to UUID.\n");
		goto EXIT;
	}

	gattlib_register_notification(gatt_connection, notification_handler, NULL);

	ret = gattlib_notification_start(gatt_connection, &temp_uuid);
	if (ret != 0) {
		fprintf(stderr, "Fail to start notification.\n");
		goto EXIT;
	}

	puts("---- In notif ----");

	GMainLoop *loop = g_main_loop_new(NULL, 0);
	g_timeout_add_seconds(60 /*timeout*/, stop_scan_func2, loop);
	g_main_loop_run(loop);

	ret = gattlib_notification_stop(gatt_connection, &temp_uuid);
	if (ret != 0) {
		fprintf(stderr, "Fail to stop notification.\n");
		goto EXIT;
	}

	puts("---- Stop notif ----");

	GMainLoop *loop2 = g_main_loop_new(NULL, 0);
	g_timeout_add_seconds(60 /*timeout*/, stop_scan_func2, loop2);
	g_main_loop_run(loop2);

	gattlib_disconnect(gatt_connection);

```

with:

```
void notification_handler(const uuid_t* uuid, const uint8_t* data, size_t data_length, void* user_data) {
	puts("Notification!");
}
```

closed 2020-04-07 by oliviermartin

oliviermartin, 2020-04-08:
Actually, someone reported me another [issue](https://github.com/labapart/gattlib/issues/132) (with a test file). His source file highlighted the issue you mentioned.

So it should now correctly be fixed by this PR: https://github.com/labapart/gattlib/pull/163