labapart/gattlib #144: notification_handler stop receiving data after 30 seconds

caferrick, 2020-01-28:
I am sure that it is something that I am missing when I connect to heart rate Polar H7 with the notification example everything works fine. But after 30 seconds i no longer receive data. Any guidance would appreciated.

5frank, 2020-01-29:
Sounds like a similar problem I had. It was because of rejected `LE L2CAP: Connection Parameter Update Request (0x12)` and could be fixed by changing `/sys/kernel/debug/bluetooth/hci0/conn_<max/min>_interval`
You can confirm this with `btmon`  (In my case the bluez dbus API did not provide any info on why it was disconnected)

caferrick, 2020-01-30:
Thanks you very much for the quick response. I tried larger values smaller
values to no avail. The bluetooth 5 chip has an issue. Looks like i have to
build my prototype on Pi 3 it work without any issues. Maybe have to look
at the googles new board as well. Such a shame had such high hopes 4gb ram
the device was super responsive. I may just return it I have another week
or so.

Thanks again for your help
Chuck

caferrick, 2020-01-30:
Hello,

   I resolved the issue with a full upgrade to the OS. I was running the
Raspbian image form 9.26.2019 and ran the sudo apt-get upgrade. I though
for sure I did this step right after i booted the first time bust not off.
Thanks again for your help

oliviermartin, 2020-04-07:
Thanks for having tell me the issue is solved. I close the bug.

closed 2020-04-07 by oliviermartin