labapart/gattlib #148: Build Problem

eren23, 2020-03-17:
Hi,
This the error I get when I try to run cmake .. command
It's running on raspbian.
Thanks,

-- Checking for one of the modules 'glib-2.0'
CMake Error at /usr/local/share/cmake-3.4/Modules/FindPkgConfig.cmake:568 (message):
  None of the required 'glib-2.0' found
Call Stack (most recent call first):
  dbus/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'gio-unix-2.0'
CMake Error at /usr/local/share/cmake-3.4/Modules/FindPkgConfig.cmake:568 (message):
  None of the required 'gio-unix-2.0' found
Call Stack (most recent call first):
  dbus/CMakeLists.txt:30 (pkg_search_module)

Build DBus gattlib for Bluez v5.50
CMake Error at /usr/local/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.4/Modules/FindDoxygen.cmake:107 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  docs/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/Desktop/gattlib/build/CMakeFiles/CMakeOutput.log".

dumanovskis, 2020-03-18:
Try not building any docs. See #147.

Raggyman, 2020-03-18:
Got the same error, and ran it with the no docs and got 

`root@PPD100:/home/pi/gattlib/build# cmake -DGATTLIB_BUILD_DOCS=NO ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for one of the modules 'bluez'
-- Checking for one of the modules 'glib-2.0'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'glib-2.0' found
Call Stack (most recent call first):
  dbus/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'gio-unix-2.0'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gio-unix-2.0' found
Call Stack (most recent call first):
  dbus/CMakeLists.txt:30 (pkg_search_module)

Build DBus gattlib for Bluez v5.43
-- Checking for one of the modules 'python3'
-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/ble_scan/CMakeLists.txt:26 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/ble_scan/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/discover/CMakeLists.txt:26 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/discover/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/read_write/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/read_write/CMakeLists.txt:28 (pkg_search_module)

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/read_write_memory/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'glib-2.0'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'glib-2.0' found
Call Stack (most recent call first):
  examples/read_write_memory/CMakeLists.txt:28 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/read_write_memory/CMakeLists.txt:29 (pkg_search_module)

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/notification/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'glib-2.0'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'glib-2.0' found
Call Stack (most recent call first):
  examples/notification/CMakeLists.txt:28 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/notification/CMakeLists.txt:29 (pkg_search_module)

-- Checking for one of the modules 'gattlib'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'gattlib' found
Call Stack (most recent call first):
  examples/nordic_uart/CMakeLists.txt:27 (pkg_search_module)

-- Checking for one of the modules 'libpcre'
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:637 (message):
  None of the required 'libpcre' found
Call Stack (most recent call first):
  examples/nordic_uart/CMakeLists.txt:28 (pkg_search_module)

Package Gattlib for development version
-- Configuring incomplete, errors occurred!
See also "/home/pi/gattlib/build/CMakeFiles/CMakeOutput.log".`

oliviermartin, 2020-04-03:
@Raggyman Install `libpcre3-dev` (`apt install libpcre3-dev`) and try again.

PR #153 (2020-04-03): doc: Check if Doxygen is installed before generating the doc
Fix #148
closed 2020-04-03 by oliviermartin