labapart/gattlib #151: Can't unregister indications

whaatt, 2020-04-01:
I'm having the same issue as #140, where it looks like the address value of `dbus_characteristic.gatt` is changing between `gattlib_indication_start` and `gattlib_indication_stop`, such that I can't unregister from an indication with some UUID and re-register to it without getting duplicate indications.

Thanks in advance for the help!

oliviermartin, 2020-04-08:
@whaatt when you say you have an issue, does the function returns an error or you still see indication? Which OS and version of Bluez are you using?

oliviermartin, 2020-04-08:
Actually, someone reported me another [issue](https://github.com/labapart/gattlib/issues/132) (with a test file). His source file highlighted the issue you mentioned.

So it should now correctly be fixed by this PR: https://github.com/labapart/gattlib/pull/163

closed 2020-04-08 by oliviermartin