labapart/gattlib #156: How to scan for Eddystone beacons?

jsochurek, 2020-04-06:
Hi I was able to get the `ble_scan` example to run and detect devices. I was hoping to see if i could modify it to scan for Eddystone only. I tried this code below, but I don't get any discoveries. I am curious if this is possible in this manor, or I am off base? I've tried messing around with the rssi threshold as well. Thanks!

`
#include <pthread.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <sys/queue.h>

#include "gattlib.h"

#define BLE_SCAN_TIMEOUT   10

typedef void (*ble_discovered_device_t)(const char* addr, const char* name);

// We use a mutex to make the BLE connections synchronous
static pthread_mutex_t g_mutex = PTHREAD_MUTEX_INITIALIZER;

LIST_HEAD(listhead, connection_t) g_ble_connections;
struct connection_t {
	pthread_t thread;
	char* addr;
	LIST_ENTRY(connection_t) entries;
};

static void *ble_connect_device(void *arg) {
	struct connection_t *connection = arg;
	char* addr = connection->addr;
	gatt_connection_t* gatt_connection;
	gattlib_primary_service_t* services;
	gattlib_characteristic_t* characteristics;
	int services_count, characteristics_count;
	char uuid_str[MAX_LEN_UUID_STR + 1];
	int ret, i;

	pthread_mutex_lock(&g_mutex);

	printf("------------START %s ---------------\n", addr);

	gatt_connection = gattlib_connect(NULL, addr, GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT);
	if (gatt_connection == NULL) {
		fprintf(stderr, "Fail to connect to the bluetooth device.\n");
		goto connection_exit;
	} else {
		puts("Succeeded to connect to the bluetooth device.");
	}

	ret = gattlib_discover_primary(gatt_connection, &services, &services_count);
	if (ret != 0) {
		fprintf(stderr, "Fail to discover primary services.\n");
		goto disconnect_exit;
	}

	for (i = 0; i < services_count; i++) {
		gattlib_uuid_to_string(&services[i].uuid, uuid_str, sizeof(uuid_str));

		printf("service[%d] start_handle:%02x end_handle:%02x uuid:%s\n", i,
				services[i].attr_handle_start, services[i].attr_handle_end,
				uuid_str);
	}
	free(services);

	ret = gattlib_discover_char(gatt_connection, &characteristics, &characteristics_count);
	if (ret != 0) {
		fprintf(stderr, "Fail to discover characteristics.\n");
		goto disconnect_exit;
	}
	for (i = 0; i < characteristics_count; i++) {
		gattlib_uuid_to_string(&characteristics[i].uuid, uuid_str, sizeof(uuid_str));

		printf("characteristic[%d] properties:%02x value_handle:%04x uuid:%s\n", i,
				characteristics[i].properties, characteristics[i].value_handle,
				uuid_str);
	}
	free(characteristics);

disconnect_exit:
	gattlib_disconnect(gatt_connection);

connection_exit:
	printf("------------DONE %s ---------------\n", addr);
	pthread_mutex_unlock(&g_mutex);
	return NULL;
}

// static void ble_discovered_device(void *adapter, const char* addr, const char* name, void *user_data) {
// 	struct connection_t *connection;
// 	int ret;

// 	if (name) {
// 		printf("Discovered %s - '%s'\n", addr, name);
// 	} else {
// 		printf("Discovered %s\n", addr);
// 	}

// 	connection = malloc(sizeof(struct connection_t));
// 	if (connection == NULL) {
// 		fprintf(stderr, "Failt to allocate connection.\n");
// 		return;
// 	}
// 	connection->addr = strdup(addr);

// 	ret = pthread_create(&connection->thread, NULL,	ble_connect_device, connection);
// 	if (ret != 0) {
// 		fprintf(stderr, "Failt to create BLE connection thread.\n");
// 		free(connection);
// 		return;
// 	}
// 	LIST_INSERT_HEAD(&g_ble_connections, connection, entries);
// }

static void ble_discovered_eddystone_device(void *adapter, const char* addr, const char* name,
		gattlib_advertisement_data_t *advertisement_data, size_t advertisement_data_count,
		uint16_t manufacturer_id, uint8_t *manufacturer_data, size_t manufacturer_data_size,
		void *user_data) {
    struct connection_t *connection;
    int ret;

    if (name) {
        printf("Discovered %s - '%s'\n", addr, name);
    } else {
        printf("Discovered %s\n", addr);
    }

    connection = malloc(sizeof(struct connection_t));
    if (connection == NULL) {
        fprintf(stderr, "Failt to allocate connection.\n");
        return;
    }
    connection->addr = strdup(addr);

    ret = pthread_create(&connection->thread, NULL,	ble_connect_device, connection);
    if (ret != 0) {
        fprintf(stderr, "Failt to create BLE connection thread.\n");
        free(connection);
        return;
    }
    LIST_INSERT_HEAD(&g_ble_connections, connection, entries);
}

int main(int argc, const char *argv[]) {
	const char* adapter_name;
	void* adapter;
	int ret;

	if (argc == 1) {
		adapter_name = NULL;
	} else if (argc == 2) {
		adapter_name = argv[1];
	} else {
		fprintf(stderr, "%s [<bluetooth-adapter>]\n", argv[0]);
		return 1;
	}

	LIST_INIT(&g_ble_connections);

	ret = gattlib_adapter_open(adapter_name, &adapter);
	if (ret) {
		fprintf(stderr, "ERROR: Failed to open adapter.\n");
		return 1;
	}

	pthread_mutex_lock(&g_mutex);
	int rssi = 0;
    ret = gattlib_adapter_scan_eddystone(adapter, rssi, GATTLIB_EDDYSTONE_TYPE_URL, ble_discovered_eddystone_device, BLE_SCAN_TIMEOUT, NULL );
	if (ret) {
		fprintf(stderr, "ERROR: Failed to scan.\n");
		goto EXIT;
	}

	gattlib_adapter_scan_disable(adapter);

	puts("Scan completed");
	pthread_mutex_unlock(&g_mutex);

	// Wait for the thread to complete
	while (g_ble_connections.lh_first != NULL) {
		struct connection_t* connection = g_ble_connections.lh_first;
		pthread_join(connection->thread, NULL);
		LIST_REMOVE(g_ble_connections.lh_first, entries);
		free(connection->addr);
		free(connection);
	}

EXIT:
	gattlib_adapter_close(adapter);
	return ret;
}
`

oliviermartin, 2020-04-06:
Hi @jsochurek , actually gattlib has already a function to scan for Eddystone, see [gattlib_adapter_scan_eddystone()](https://github.com/labapart/gattlib/blob/master/include/gattlib.h#L248)

```
/**
 * @name Gattlib Eddystone types
 */
//@{
#define GATTLIB_EDDYSTONE_TYPE_UID                          (1 << 0)
#define GATTLIB_EDDYSTONE_TYPE_URL                          (1 << 1)
#define GATTLIB_EDDYSTONE_TYPE_TLM                          (1 << 2)
#define GATTLIB_EDDYSTONE_TYPE_EID                          (1 << 3)
#define GATTLIB_EDDYSTONE_LIMIT_RSSI                        (1 << 4)
//@}

/**
 * @brief Enable Eddystone Bluetooth Device scanning on a given adapter
 *
 * @param adapter is the context of the newly opened adapter
 * @param rssi_threshold is the imposed RSSI threshold for the returned devices.
 * @param eddystone_types defines the type(s) of Eddystone advertisement data type to select.
 *        The types are defined by the macros `GATTLIB_EDDYSTONE_TYPE_*`. The macro `GATTLIB_EDDYSTONE_LIMIT_RSSI`
 *        can also be used to limit RSSI with rssi_threshold.
 * @param discovered_device_cb is the function callback called for each new Bluetooth device discovered
 * @param timeout defines the duration of the Bluetooth scanning
 * @param user_data is the data passed to the callback `discovered_device_cb()`
 *
 * @return GATTLIB_SUCCESS on success or GATTLIB_* error code
 */
int gattlib_adapter_scan_eddystone(void *adapter, int16_t rssi_threshold, uint32_t eddystone_types,
		gattlib_discovered_device_with_data_t discovered_device_cb, int timeout, void *user_data);
```

You can also have a look to this example to see how to use it (in python, but Gattlib python API mirror the C API - and actually uses the C API for backend): https://github.com/labapart/gattlib/blob/master/gattlib-py/examples/find_eddystone/find_eddystone.py

jsochurek, 2020-04-06:
Yes I tried to use that but I'm not getting any devices discovered. I've used other apps to verify that my beacons are broadcasting Eddystone UID. I was hoping you could tell me if my usage of the function above was incorrect

oliviermartin, 2020-04-06:
Arg sorry, I read your message too quickly. I might give a try to see if I found the issue in your code tomorrow.

oliviermartin, 2020-04-07:
I have created a new example to demonstrate [scanning of Eddystone beacon](https://github.com/labapart/gattlib/blob/master/examples/find_eddystone/find_eddystone.c). I have tested with [Nordic Thingy](https://www.nordicsemi.com/Software-and-tools/Prototyping-platforms/Nordic-Thingy-52) and it works fine for me.

Let me know if you still have issue.

jsochurek, 2020-04-07:
This worked for me! Thanks so much. I think my issue was, i was using the release of the library. But that appears to be out of date. After building the lib and installing that way, I'm able to detect UID and URL advertisements (I added `GATTLIB_EDDYSTONE_TYPE_UID` to the scan filter)

closed 2020-04-07 by jsochurek