labapart/gattlib #165: Advertisement data update frequency

z3ntu, 2020-04-08:
Hi.
I have a BLE device which sends interesting data as manufacturer data in the advertisement packets which I want to get and print. As it updates frequently I want fresh data at least once a second.

I've made a simple test application for reading the advertisement data, basically a bastardized version of `gattlib-py/examples/ble_scan/ble_scan.py` which has the following as `connect_ble_device` (uses #164):
```python
def connect_ble_device(device):
    device.connect()
    while True:
        data = device.get_advertisement_data()
        print(data)
        time.sleep(0.5)
```
What I'm seeing is that the data returned from isn't really updated at all after starting the application and getting the first advertisement data. Do you know what I'm doing wrong or how it should be done to achieve my goal?
Thanks a lot for this library, it seems to be one of the few BLE libraries for Linux which doesn't use deprecated tools or isn't maintained anymore!

oliviermartin, 2020-04-08:
It might also be a bug in gattlib - I am not sure. Looking quickly at the code of `get_advertisement_data()`, it seems it could work - but maybe not.

To investigate (I am not sure I have such BLE device at home that periodically sends changing advertisement data), use `bluetoothctl` in command line and use the command `scan on`. Confirm `bluetoothctl` see the advertisement data change (I would expect to see a line such as `[CHG] ....` in your terminal).

oliviermartin, 2020-04-09:
Actually, i thought about your issue last night. And I think it comes from the fact you are connected to the device. And often BLE advertising is stopped when connecting to the device - so no refreshment of the advertisement data.

You are even luckier! I was hearing this morning on the radio about the [European Project "Stop Covid"](https://techcrunch.com/2020/04/08/france-is-officially-working-on-stop-covid-contact-tracing-app/). That will use BLE technology.

And I was thinking how it could work. And I am guessing it uses Bluetooth Advertisement Data.

So I added support to Gattlib to fix your issue - and it could potentially used for any "Stop Covid" initiative...

Here is the example: https://github.com/labapart/gattlib/blob/master/gattlib-py/examples/advertisement_data/advertisement_data.py

Let me know if it fixes your issue. I close the issue for now.

closed 2020-04-09 by oliviermartin

z3ntu, 2020-04-09:
Thanks for implementing this so quickly! With this function I get new data into the on_discovered_ble_device function roughly (but pretty consistently) every 16 seconds and the data is up-to-date, so it's already much better! Looking the bluetoothctl output, it's the same rate I get data there as in gattlib.
I still wonder why I can't get advertisement every second (or even quicker) into my application though and where the "bottleneck" lies (or where this bottleneck can be configured).
According to the "nRF Connect" app, the BLE device sends data roughtly every 20ms 
[image]

oliviermartin, 2020-04-09:
Do you know if the advertisement data are actually different every 20ms?

z3ntu, 2020-04-10:
Maybe not every 20ms but the advertised data should be updated at least once per second as the data itself contains a timer in seconds which increments.

oliviermartin, 2020-04-14:
@z3ntu If you want to go know more about this specific issue, I would recommend you to ask your question on Bluez mailing-list: http://www.bluez.org/development/lists/

I would actually be curious to know the answer to know if there is something I can do with gattlib :-)

z3ntu, 2020-05-07:
Thanks for the tip. I searched the mailing list archive a bit today and this which looks interesting: https://lore.kernel.org/linux-bluetooth/20200501161017.BlueZ.v5.1.I137a529ab03c9d0d2327f1659bd1af4954a28e78@changeid/ 
I couldn't find any actual code for this doc proposal though
```
This API allows an client to specify a job of monitoring advertisements by
registering the root of hierarchy and then exposing advertisement monitors
under the root with filtering conditions, thresholds of RSSI and timers
of RSSI thresholds.

Once a monitoring job is activated by BlueZ, the client can expect to get
notified on the targeted advertisements no matter if there is an ongoing
discovery session (a discovery session is started/stopped with methods in
org.bluez.Adapter1 interface).
```

I've also sent an email with my specific question to the list, hopefully someone can help :)