labapart/gattlib #168: Segfault in bt_uuid_to_uuid

z3ntu, 2020-04-09:
Running the new advertisement_data example, it crashes with the following backtrace:
```
Thread 1 "advertisement_d" received signal SIGSEGV, Segmentation fault.
bt_uuid_to_uuid (bt_uuid=0x7fffffffd1d0, uuid=0x2) at /tmp/gattlib/common/gattlib_common.c:102
102             memcpy(&uuid->value, &bt_uuid->value, sizeof(uuid->value));
(gdb) bt
#0  bt_uuid_to_uuid (bt_uuid=0x7fffffffd1d0, uuid=0x2) at /tmp/gattlib/common/gattlib_common.c:102
#1  0x00007ffff7f9f70a in gattlib_string_to_uuid (str=0x55555559e418 "00002018-0000-1000-8000-00805f9b34fb", n=36, uuid=0x2) at /tmp/gattlib/common/gattlib_common.c:149
#2  0x00007ffff7f9cb56 in get_advertisement_data_from_device
    (bluez_device1=0x55555556eb60, advertisement_data=0x7fffffffd360, advertisement_data_count=0x7fffffffd368, manufacturer_id=0x7fffffffd35a, manufacturer_data=0x7fffffffd370, manufacturer_data_size=0x7fffffffd378) at /tmp/gattlib/dbus/gattlib_advertisement.c:103
#3  0x00007ffff7f9cd9e in gattlib_get_advertisement_data_from_mac
    (adapter=0x555555586980, mac_address=0x7fffe802b500 "18:E8:29:B8:1E:B8", advertisement_data=0x7fffffffd360, advertisement_data_count=0x7fffffffd368, manufacturer_id=0x7fffffffd35a, manufacturer_data=0x7fffffffd370, manufacturer_data_size=0x7fffffffd378) at /tmp/gattlib/dbus/gattlib_advertisement.c:159
#4  0x0000555555555229 in ble_advertising_device (adapter=0x555555586980, addr=0x7fffe802b500 "18:E8:29:B8:1E:B8", name=0x7fffe8028880 "UCK", user_data=0x0)
    at /tmp/gattlib/examples/advertisement_data/advertisement_data.c:17
#5  0x00007ffff7f9c10f in device_manager_on_device1_signal (device1_path=0x555555593070 "/org/bluez/hci0/dev_18_E8_29_B8_1E_B8", arg=0x7fffffffda20) at /tmp/gattlib/dbus/gattlib_adapter.c:158
#6  0x00007ffff7f9c218 in on_interface_proxy_properties_changed
    (device_manager=0x55555557a9d0, object_proxy=0x555555566b20, interface_proxy=0x5555555911a0, changed_properties=0x7fffe802a690, invalidated_properties=0x55555559e3c0, user_data=0x7fffffffda20) at /tmp/gattlib/dbus/gattlib_adapter.c:198
#7  0x00007ffff7d76b48 in  () at /usr/lib/libgio-2.0.so.0
#8  0x00007ffff7c4a0a0 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#9  0x00007ffff7c4b6b0 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#10 0x00007ffff7cd803c in  () at /usr/lib/libgio-2.0.so.0
#11 0x00007ffff7cf2794 in  () at /usr/lib/libgio-2.0.so.0
#12 0x00007ffff7ea488f in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#13 0x00007ffff7ea6831 in  () at /usr/lib/libglib-2.0.so.0
#14 0x00007ffff7ea7843 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#15 0x00007ffff7f9c67a in gattlib_adapter_scan_enable_with_filter
    (adapter=0x555555586980, uuid_list=0x0, rssi_threshold=0, enabled_filters=4, discovered_device_cb=0x5555555551d9 <ble_advertising_device>, timeout=0, user_data=0x0)
    at /tmp/gattlib/dbus/gattlib_adapter.c:291
#16 0x00005555555553b9 in main (argc=1, argv=0x7fffffffdcd8) at /tmp/gattlib/examples/advertisement_data/advertisement_data.c:56
```
It's probably some device around me causing this crash because I see it every time I start the example.

oliviermartin, 2020-04-09:
@z3ntu , could you add in the function [gattlib_string_to_uuid()](https://github.com/labapart/gattlib/blob/master/common/gattlib_common.c#L144), the line `printf("uuid[%d]:%s\n", n, str);` and send me the output.

z3ntu, 2020-04-09:
Sure, here's the output:
```
uuid[36]:0000252a-0000-1000-8000-00805f9b34fb
uuid[36]:00002018-0000-1000-8000-00805f9b34fb
fish: “./build/examples/advertisement_…” terminated by signal SIGSEGV (Address boundary error)
```

z3ntu, 2020-04-09:
So I've looked at the code now and with my limited C and double pointer knowledge, I think we have the following situation
```
// definition of advertisement_data
..., gattlib_advertisement_data_t **advertisement_data, ...

// advertisement_data_count is 2 in this case

// allocate memory
*advertisement_data = calloc(sizeof(gattlib_advertisement_data_t), *advertisement_data_count);

// accessing advertisement_data[0] results in a valid pointer e.g. 0x556d90160b70
// accessing advertisement_data[1] results in an invalid pointer 0x2
```
What I'm guessing is that `*advertisement_data` is initialized with the pointer to the allocated memory but we're accessing the next pointer (`advertisement_data[1]`) which points to something else

z3ntu, 2020-04-09:
It seems that with this diff it's compiling and it isn't crashing anymore :)
```diff
diff --git a/dbus/gattlib_advertisement.c b/dbus/gattlib_advertisement.c
index 3fa7040..f6d1ea0 100644
--- a/dbus/gattlib_advertisement.c
+++ b/dbus/gattlib_advertisement.c
@@ -100,20 +100,20 @@ int get_advertisement_data_from_device(OrgBluezDevice1 *bluez_device1,
 
 		g_variant_get(service_data_variant, "a{sv}", &iter);
 		while (g_variant_iter_loop(iter, "{&sv}", &key, &value)) {
-			gattlib_string_to_uuid(key, strlen(key), &advertisement_data[index]->uuid);
+			gattlib_string_to_uuid(key, strlen(key), &(&(*advertisement_data)[index])->uuid);
 
 			gsize n_elements = 0;
 			gconstpointer const_buffer = g_variant_get_fixed_array(value, &n_elements, sizeof(guchar));
 			if (const_buffer) {
-				advertisement_data[index]->data = malloc(n_elements);
-				if (advertisement_data[index]->data == NULL) {
+				(&(*advertisement_data)[index])->data = malloc(n_elements);
+				if ((&(*advertisement_data)[index])->data == NULL) {
 					return GATTLIB_OUT_OF_MEMORY;
 				}
 
-				advertisement_data[index]->data_length = n_elements;
-				memcpy(advertisement_data[index]->data, const_buffer, n_elements);
+				(&(*advertisement_data)[index])->data_length = n_elements;
+				memcpy((&(*advertisement_data)[index])->data, const_buffer, n_elements);
 			} else {
-				advertisement_data[index]->data_length = 0;
+				(&(*advertisement_data)[index])->data_length = 0;
 			}
 
 			index++;

```
It seems to be not completely incorrect as it's not crashing :wink:

oliviermartin, 2020-04-09:
Thanks for having investigating the issue. You are right, there was a wrong access to the allocated structure.

closed 2020-04-09 by oliviermartin

z3ntu, 2020-04-10:
Thanks, your fix works fine (and looks more elegant :slightly_smiling_face: )!

PR #169 (2020-04-09): dbus/gattlib_advertisement: Fix advertisement_data access in get_advertisement_data_from_device()
Fix: #168