labapart/gattlib #170: Disconnection Event

tiagotmg, 2020-04-15:
Hello,
I'm using gattlib to connect with a BLE peripheral. Is there a way to get an event if the device disconnects? I tried with `gattlib_register_on_disconnect()` but it doesn't work when it is the peripheral who drops the connection.

Any idea/suggestion is appreciated!

Thank you

tiagotmg, 2020-04-22:
I was using a Linux virtual machine with Virtual Box. When I tested exactly the same code with a real machine, it worked well. So, not a library problem.

thanks

closed 2020-04-22 by tiagotmg

oliviermartin, 2020-04-22:
Thanks a lot @tiagotmg for having updated this ticket :-)

SONACHAI1234, 2020-10-27:
I tried with different changes but I don't know why when BLE peripheral device disconnected, the central is not getting acknowledge or the handler is not calling.
For me this is not working. I don't know we I am making mistakes can you guide me?

Here is my code:
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include "gattlib.h"

void disconnectHandler(void *arg)
{
        printf("in disconnection handler \n");
}
int main(int argc, char *argv[])
{
        gatt_connection_t* connection;
        gattlib_primary_service_t* services;
        gattlib_characteristic_t* characteristics;
        int services_count, characteristics_count;
        char uuid_str[MAX_LEN_UUID_STR + 1];
        int ret, i;

        if (argc != 2) {
                printf("%s <device_address>\n", argv[0]);
                return 1;
        }

        connection = gattlib_connect(NULL, argv[1], GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT);
        if (connection == NULL) {
                fprintf(stderr, "Fail to connect to the bluetooth device.\n");
                return 1;
        }
        if(gattlib_has_valid_handler(disconnectHandler) == 1)
        {
                printf("handler is valid\n");
                gattlib_register_on_disconnect(connection, disconnectHandler, NULL);
        }
        else
        {
                printf("handler is not valid\n");
        }
        ret = gattlib_discover_primary(connection, &services, &services_count);
        if (ret != GATTLIB_SUCCESS) {
                fprintf(stderr, "Fail to discover primary services.\n");
                return 1;
        }

        for (i = 0; i < services_count; i++) {
                gattlib_uuid_to_string(&services[i].uuid, uuid_str, sizeof(uuid_str));

                printf("service[%d] start_handle:%02x end_handle:%02x uuid:%s\n", i,
                                services[i].attr_handle_start, services[i].attr_handle_end,
                                uuid_str);
        }
        free(services);

        ret = gattlib_discover_char(connection, &characteristics, &characteristics_count);
        if (ret != GATTLIB_SUCCESS) {
                fprintf(stderr, "Fail to discover characteristics.\n");
                return 1;
        }
        for (i = 0; i < characteristics_count; i++) {
                gattlib_uuid_to_string(&characteristics[i].uuid, uuid_str, sizeof(uuid_str));

                printf("characteristic[%d] properties:%02x value_handle:%04x uuid:%s\n", i,
                                characteristics[i].properties, characteristics[i].value_handle,
                                uuid_str);
        }
        free(characteristics);
        for(int i=0; i<100; i++)
        {
                sleep(1);
                printf("In loop\n");
        }

        gattlib_disconnect(connection);
        return 0;
}
-------------------------------------

In 100 seconds sleep part, I am disconnecting peripheral but I am not getting disconnectionHandler call. Am I making any basic mistakes or any bug in gettlib library? I check with btmon log that disconnection print coming but it is not riching to gattlib library or my application.

btmod logs:
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8         {0x0002} [hci0] 327.987207
        LE Address: 30:AE:A4:F5:FF:1E (Espressif Inc.)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8         {0x0001} [hci0] 327.987207
        LE Address: 30:AE:A4:F5:FF:1E (Espressif Inc.)
        Reason: Connection timeout (0x01)

I am using 21 Jul 2020 means the latest commit for gattlib library.