labapart/gattlib #177: static building dbus CMakeFile.txt error and compiling examples against static gattlib

saad-s, 2020-06-24:
When building with 
`cmake -DGATTLIB_SHARED_LIB=NO ..`
gives this error
```
CMake Error at dbus/CMakeLists.txt:127 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "gattlib".
```
modifying dbus/CMakeLists.txt;127 to this solves the error and generates libgattlib.a
```
if(GATTLIB_SHARED_LIB)
  install(TARGETS gattlib LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR})
else()
  install(TARGETS gattlib ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR})
endif()
```
I added following in ble_scan example's CMakeLists.txt for static building 
```
SET(CMAKE_FIND_LIBRARY_SUFFIXES ".a")
SET(BUILD_SHARED_LIBS OFF)
SET(CMAKE_EXE_LINKER_FLAGS "-static")
```
and it gives undefined reference to... errors, I've added all dev libs like gio-2.0, gobject etc and it reduced errors but now its asking for libsystem and libmount etc..

My question is, is there a proper way to build static executable so I can move it to my docker image without installing dev libs in my image ?

saad-s, 2020-07-05:
adding this list of libs generates a statically linked binary / executable 
```
-lgattlib \
-lglib-2.0 -lpcre \
-ldbus-1 -lpthread \
-lgio-2.0 -lz -lresolv -lselinux -lmount -lgmodule-2.0  -ldl -lgobject-2.0 -lffi -lglib-2.0 -lpcre \
-lgmodule-2.0 -ldl -lglib-2.0 -lpcre \
-lgobject-2.0 -lffi -lglib-2.0 -lpcre \
-ldbus-glib-1 -ldbus-1 -lgobject-2.0 -lffi -lglib-2.0 -lpcre -lmount -lblkid -luuid 
```
a better way could be to link statically against specific dev libs and dynamically link against common libs using 
```
-Wl,-Bstatic and -Wl,-Bdynamic
```
but that's a task for some other day.

close the issue please

alepar, 2022-01-29:
Thank you, the above worked, just had to add `-lpython3.9` to the list.