labapart/gattlib #188: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort error received while trying to establish connection with BLE device.

hizzawaseem, 2020-08-20:
I am facing GDBus error while establishing connection with the BLE device frequently via Raspberry Pi 3. Mostly connection establishes successfully but every now and then the following error appears, and the code exits. The error message is:

**Device connected error (device:/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX): GDBus.Error:org.bluez.Error.Failed: Software caused connection abort**

Interface is initialized properly and after successful connection and data sharing, we close the connection gracefully as well. 
The command that I am using to establish connection to the BLE Device is: 

**gatt_connection = gattlib_connect(NULL, mac, GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT);**

I am running my code using a docker container using following docker run command: 

docker run --name lockers --rm --privileged --net=host -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket -v /dev:/dev -v /home/pi/config:/root/config -v /etc/ssl/certs:/etc/ssl/certs:ro -v /etc/localtime:/etc/localtime:ro mizumono/smartlockers:latest --wireless --config /root/config/config.json --logdir /root/config/

ForarestLiu, 2021-06-10:
I have the same problem as you. Have you resolved this problem? Thank you.

Randomfinger, 2022-02-10:
This solved my "GDBus.Error:org.bluez.Error" .. "connection abort":
`
sudo apt install pulseaudio-module-bluetooth &
pulseaudio -k & pulseaudio --start
`