labapart/gattlib #202: Memory leak issue with gattlib and glib

ravi-vt, 2021-02-26:
I have found some memory leaks with gattlib and glib.

For gattlib, I am getting a memory leak from "gattlib_adapter_open".
I observed leaks in glib also. this block leaking too much memory that we can't afford.
I found these memory leaks using Valgrind. its log is as below.

==2700== 200 (160 direct, 40 indirect) bytes in 8 blocks are definitely lost in loss record 11 of 31
==2700==    at 0x4842590: calloc (vg_replace_malloc.c:752)
==2700==    by 0x4A85BB7: gattlib_adapter_open (in /usr/lib/libgattlib.so)

==2700== 29,288 (7,840 direct, 21,448 indirect) bytes in 280 blocks are definitely lost in loss record 27 of 31
==2700==    at 0x483F970: malloc (vg_replace_malloc.c:299)
==2700==    by 0x49D9E7B: g_malloc (in /usr/lib/libglib-2.0.so.0.5800.3)

Please provide some patches or solutions to resolve the leak issue.

Thank you.

vl-80, 2022-02-21:
@ravi-vt 

If you are still experiencing this problem try this patch and see if the memory is still leaking:

https://github.com/vl-80/gattlib/commit/a0f68ea4c67e8063ae12d1c6efd3c5e8b285779d

vl-80, 2022-02-23:
@oliviermartin,

Can you please check this diff: https://github.com/vl-80/gattlib/commit/a0f68ea4c67e8063ae12d1c6efd3c5e8b285779d and see if it is necessary to unref the `conn_context->adapter->adapter_proxy` in gattlib_disconnect()?

It is allocated here:
https://github.com/labapart/gattlib/blob/2448da5410e5f9306fa446a8dcce99ff0877d283/dbus/gattlib_adapter.c#L26-L30

And it appears that it is not freed anywhere else.