labapart/gattlib #228: Connecting taking long time...

nsnnsn, 2021-08-29:
When i use connection = gattlib_connect(NULL, address, GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT); it takes about 4 seconds to connect. When i use the bluetoothctl utility it takes less than 2 seconds.

time bluetoothctl connect XX:XX:XX:XX:XX:XX
Attempting to connect to XX:XX:XX:XX:XX:XX
[CHG] Device DA:DC:22:B3:94:93 Connected: yes
Connection successful
real    0m 1.56s
user    0m 0.07s
sys     0m 0.02s

How can i speed it up ?
Rest is working ok...

felwiz, 2021-11-05:
I also am having this problem. Have you found a solution?

SunilKS-nd, 2023-11-01:
Try changing the `gattlib_connect` options.

i.e. `GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT` expands to  
```
#define GATTLIB_CONNECTION_OPTIONS_LEGACY_DEFAULT \
		GATTLIB_CONNECTION_OPTIONS_LEGACY_BDADDR_LE_PUBLIC | \
		GATTLIB_CONNECTION_OPTIONS_LEGACY_BDADDR_LE_RANDOM | \
		GATTLIB_CONNECTION_OPTIONS_LEGACY_BT_SEC_LOW
```

Try combinations like : 
```
gattlib_connect(NULL, address,
				GATTLIB_CONNECTION_OPTIONS_LEGACY_BDADDR_LE_RANDOM |
				GATTLIB_CONNECTION_OPTIONS_LEGACY_BT_SEC_LOW);
```
Let me know if it has improved.