labapart/gattlib #256: Fatal Python error: take_gil: PyMUTEX_LOCK(gil->switch_mutex) failed

wcbonner, 2022-12-25:
Everything built on my Raspberry Pi4 running the latest Raspian `Linux WimPi4-Dev 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux` except for a couple of warning messages

```
[ 35%] Building C object dbus/CMakeFiles/gattlib.dir/__/common/gattlib_common.c.o
/home/wim/gattlib/common/gattlib_common.c: In function ‘gattlib_call_notification_handler’:
/home/wim/gattlib/common/gattlib_common.c:77:3: warning: ‘PyEval_CallObjectWithKeywords’ is deprecated [-Wdeprecated-declarations]
   77 |   PyEval_CallObject((PyObject *)handler->notification_handler, arglist);
      |   ^~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from /home/wim/gattlib/common/gattlib_common.c:8:
/usr/include/python3.9/ceval.h:17:43: note: declared here
   17 | Py_DEPRECATED(3.9) PyAPI_FUNC(PyObject *) PyEval_CallObjectWithKeywords(
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wim/gattlib/common/gattlib_common.c: In function ‘gattlib_call_disconnection_handler’:
/home/wim/gattlib/common/gattlib_common.c:98:6: warning: ‘PyEval_CallObjectWithKeywords’ is deprecated [-Wdeprecated-declarations]
   98 |      PyEval_CallObject((PyObject *)handler->disconnection_handler, arglist);
      |      ^~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from /home/wim/gattlib/common/gattlib_common.c:8:
/usr/include/python3.9/ceval.h:17:43: note: declared here
   17 | Py_DEPRECATED(3.9) PyAPI_FUNC(PyObject *) PyEval_CallObjectWithKeywords(
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 38%] Building C object dbus/CMakeFiles/gattlib.dir/__/common/gattlib_eddystone.c.o
```

When I tried to run the ble_scan I got an error message I don't understand. It wasn't fixed by running it with sudo privileges. 

```
wim@WimPi4-Dev:~/gattlib/build $ ./examples/ble_scan/ble_scan
Fatal Python error: take_gil: PyMUTEX_LOCK(gil->mutex) failed
Python runtime state: unknown

Aborted
wim@WimPi4-Dev:~/gattlib/build $ sudo ./examples/ble_scan/ble_scan
Fatal Python error: take_gil: PyMUTEX_LOCK(gil->switch_mutex) failed
Python runtime state: unknown

Aborted
```

Huangkaiping, 2023-07-04:
> Everything built on my Raspberry Pi4 running the latest Raspian `Linux WimPi4-Dev 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux` except for a couple of warning [...]

I encountered the same problem, did you solve it later？

wcbonner, 2023-07-04:
> I encountered the same problem, did you solve it later？

@Huangkaiping Sorry to say that I'd forgotten entirely about this, and no, I never came across a solution. 

I've updated my dev machine
```
wim@WimPi4-Dev:~/gattlib/build $ uname -a
Linux WimPi4-Dev 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```
but doing a git pull in the gattlib directory told me I was up to date. I reran cmake and make and the non-sudo ble_scan command no longer produced the same error, I got my BLE development done last year and am not currently thinking about these tools.

```
wim@WimPi4-Dev:~/gattlib $ rm -r build
wim@WimPi4-Dev:~/gattlib $ mkdir build && cd build
wim@WimPi4-Dev:~/gattlib/build $ cmake ..
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Checking for one of the modules 'bluez'
-- Checking for one of the modules 'glib-2.0'
-- Checking for one of the modules 'gio-unix-2.0'
Build DBus gattlib for Bluez v5.55
-- Found Python: /usr/bin/python3.9 (found version "3.9.2") found components: Interpreter Development Development.Module Development.Embed
-- Checking for one of the modules 'gattlib'
-- Checking for one of the modules 'libpcre'
Package Gattlib for development version
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wim/gattlib/build
wim@WimPi4-Dev:~/gattlib/build $ make
[  2%] Generate D-Bus 'org.bluez.Battery1.xml'
[  5%] Generate D-Bus 'org.bluez.Adapter1.xml'
[  7%] Generate D-Bus 'org.bluez.Device1.xml'
[ 10%] Generate D-Bus 'org.bluez.GattCharacteristic1.xml'
[ 12%] Generate D-Bus 'org.bluez.GattDescriptor1.xml'
[ 15%] Generate D-Bus 'org.bluez.GattService1.xml'
Scanning dependencies of target gattlib
[ 17%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib.c.o
[ 20%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib_adapter.c.o
[ 23%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib_advertisement.c.o
[ 25%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib_char.c.o
[ 28%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib_notification.c.o
[ 30%] Building C object dbus/CMakeFiles/gattlib.dir/gattlib_stream.c.o
[ 33%] Building C object dbus/CMakeFiles/gattlib.dir/bluez5/lib/uuid.c.o
[ 35%] Building C object dbus/CMakeFiles/gattlib.dir/__/common/gattlib_common.c.o
/home/wim/gattlib/common/gattlib_common.c: In function ‘gattlib_call_notification_handler’:
/home/wim/gattlib/common/gattlib_common.c:77:3: warning: ‘PyEval_CallObjectWithKeywords’ is deprecated [-Wdeprecated-declarations]
   77 |   PyEval_CallObject((PyObject *)handler->notification_handler, arglist);
      |   ^~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from /home/wim/gattlib/common/gattlib_common.c:8:
/usr/include/python3.9/ceval.h:17:43: note: declared here
   17 | Py_DEPRECATED(3.9) PyAPI_FUNC(PyObject *) PyEval_CallObjectWithKeywords(
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wim/gattlib/common/gattlib_common.c: In function ‘gattlib_call_disconnection_handler’:
/home/wim/gattlib/common/gattlib_common.c:98:6: warning: ‘PyEval_CallObjectWithKeywords’ is deprecated [-Wdeprecated-declarations]
   98 |      PyEval_CallObject((PyObject *)handler->disconnection_handler, arglist);
      |      ^~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
[... 23 lines cut ...]
[ 71%] Building C object examples/discover/CMakeFiles/discover.dir/discover.c.o
[ 74%] Linking C executable discover
[ 74%] Built target discover
Scanning dependencies of target find_eddystone
[ 76%] Building C object examples/find_eddystone/CMakeFiles/find_eddystone.dir/find_eddystone.c.o
[ 79%] Linking C executable find_eddystone
[ 79%] Built target find_eddystone
Scanning dependencies of target read_write
[ 82%] Building C object examples/read_write/CMakeFiles/read_write.dir/read_write.c.o
[ 84%] Linking C executable read_write
[ 84%] Built target read_write
Scanning dependencies of target read_write_memory
[ 87%] Building C object examples/read_write_memory/CMakeFiles/read_write_memory.dir/read_write_memory.c.o
[ 89%] Linking C executable read_write_memory
[ 89%] Built target read_write_memory
Scanning dependencies of target notification
[ 92%] Building C object examples/notification/CMakeFiles/notification.dir/notification.c.o
[ 94%] Linking C executable notification
[ 94%] Built target notification
Scanning dependencies of target nordic_uart
[ 97%] Building C object examples/nordic_uart/CMakeFiles/nordic_uart.dir/nordic_uart.c.o
[100%] Linking C executable nordic_uart
[100%] Built target nordic_uart
wim@WimPi4-Dev:~/gattlib/build $ ./examples/ble_scan/ble_scan
```

closed 2023-07-04 by wcbonner