labapart/gattlib #257: read_write example throws "Bluez v5.42+ does not expose Client Characteristic..."

harlock974, 2023-07-11:
Hello,
I'm trying to test read_write example with a bluetooth multimeter.
It works fine with gatttool : writing 0x0100 value to handle 0x015 make the device send data through notification handle 0x0014 : 
```
[60:E8:5B:6E:A3:9D][LE]> connect
Attempting to connect to 60:E8:5B:6E:A3:9D
Connection successful
[60:E8:5B:6E:A3:9D][LE]> char-write-req 0x015 0100
Characteristic value was written successfully
Notification handle = 0x0014 value: d5 f0 00 0a 01 01 00 00 00 03 01 00 00 90 0d
```
Gatttool gives the table handle/uuid : 
```
[60:E8:5B:6E:A3:9D][LE]> char-desc 0x10 0xffff
handle: 0x0010, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0011, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0012, uuid: 0000fff1-0000-1000-8000-00805f9b34fb
handle: 0x0013, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0014, uuid: 0000fff2-0000-1000-8000-00805f9b34fb
handle: 0x0015, uuid: 00002902-0000-1000-8000-00805f9b34fb
```
So for handle 0x0015 I tried read_write with uuid 00002902-0000-1000-8000-00805f9b34fb, but get the following errors : 
```
$ ./read_write 60:E8:5B:6E:A3:9D write 00002902-0000-1000-8000-00805f9b34fb 0x0100
Value to write: 0x100
gattlib_read_write: Error: Bluez v5.42+ does not expose Client Characteristic Configuration Descriptor through DBUS interface
gattlib_read_write: Could not find GATT Characteristic with UUID 0x2902. You might call the program with '--gatt-discovery'.
```

The return value of gattlib_write_char_by_uuid() is 2 (GATTLIB_NOT_FOUND) and my version of bluez is 5.53.

What I am doing wrong ?

mjaafa, 2024-01-03:
Hi I do have the same problem did you find out a solution.

harlock974, 2024-01-03:
> Hi I do have the same problem did you find out a solution.

Unfortunately not. I end up reading my device by using a script with gatttool and expect-lite. Quite an obsolete way but it works.