labapart/gattlib #276: When i coross compile for arm64 platform,  it suggest that Could not find python developer package!

e665107, 2024-04-09:
Hi:
Following is my log,  can you help me?

forlinx@ubuntu:~/work/refs/gattlib/build$ cmake ..
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Build DBus gattlib for Bluez v5.50
-- Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Python3_LIBRARIES Development Development.Module Development.Embed) (found version "3.8.10")
-- Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Development.Module) 
CMake Error at dbus/CMakeLists.txt:115 (message):
  Could not find Python developer package

-- Configuring incomplete, errors occurred!

oliviermartin, 2024-04-09:
Hi, disabling Python support from your build should fix the build issue:

`cmake -DGATTLIB_PYTHON_INTERFACE=OFF ..`

oliviermartin, 2024-04-09:
`cmake -DGATTLIB_PYTHON_INTERFACE=OFF -DGATTLIB_BUILD_EXAMPLES=YES ..`

I will change the build system to disable Python and build examples by default :-)

e665107, 2024-04-09:
I have compiled the example apps! 
Here is a need! That is to receive the wearing watch's   broadcast data!  I want to use this lib to complete it!  
Can you give me a suggestion procedure?  For instance, which apis can do that! Thanks a lot!

e665107, 2024-04-10:
Hi，
After i used this api,  it prints as follow:
gattlib_advertisement_dat: Manufacturer Data with multiple children: {uint16 1048: <[byte 0x0a, 0x0a, 0x07, 0x02, 0xca, 0x00, 0x20, 0x00, 0x00, 0x00, 0x1e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]>, 4821: <[byte 0xd8, 0x94, 0x00, 0x4e]>}

What does it mean,  i have find it in the source,  and below is "return GATTLIB_NOT_SUPPORTED;" .
 Gattlib can't support this ble device?

e665107, 2024-04-10:
When I used ble app of iphone, the manufacture data is follow:
[image]

oliviermartin, 2024-04-10:
@e665107 I have just updated the repository to support multiple manufacturer data ID. I have not tested the code yet.
If it does not work, can you send me the full log to help me to fix the issue.

e665107, 2024-04-10:
I'm very glad to do that! 

    ret = gattlib_get_advertisement_data_from_mac(adapter, addr,
                                                  &advertisement_data, &advertisement_data_count,
                                                  &manufacturer_id, &manufacturer_data, &manufacturer_data_size);
After i have replaced the addr with the wearing watch's addr "D5:12:D8:94:00:4E",  run the advertisement_data ,  it prints as follow:
gattlib_advertisement_dat: Manufacturer Data with multiple children: {uint16 1048: <[byte 0x0a, 0x0a, 0x07, 0x02, 0xca, 0x00, 0x20, 0x00, 0x00, 0x00, 0x1e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]>, 4821: <[byte 0xd8, 0x94, 0x00, 0x4e]>}

The main purpose to replace the addr is that  i want to receive the only advertise data of  wearing watch,  do i modify wrong?

oliviermartin, 2024-04-10:
@e665107 I do not think you got the last version of gattlib. The last version is [880f1d2](https://github.com/labapart/gattlib/commit/880f1d2cd0bb133e0bd9dc3eee9eb63f6efc9a98).

Do you confirm you have this version when you do `git log -1`

e665107, 2024-04-10:
Following is the  newest log, i do git clone directly:
db04a0e * origin/master Introduce gattlib_connection_is_valid() to not access 'connection->device'
2d771d9 * Fix gattlib_connection_is_connected
5406a97 * More logging
4acf4aa * connection: Ensure device_object_path is not freed multiple time
f609f7d * adapter: Added support to open multiple time the same adapter
dc00902 * Log when the adapter or device has been released
cdd62f6 * Return specific error code when the adapter or device has been removed
0e34df5 * 0.6.0 Single lock for all gattlib library
014c280 * Refactor code to better separate gattlib_adapter_t, gattlib_device_t, gattlib_connection_t and their backends
22dca45 * Require a lower version of cmake (same version as Ubuntu 22.04 LTS)
5049443 * 0.5.2 Ensure the connection structure is not freed when used by the connection thread
a587aa9 * 0.5.1 gattlib_connect: Set the device has DISCONNECTED on error
67ff1de * 0.5.0 origin/latest-fixes gattlib-py/adapter: Catch error on opening/closing BLE adapter
53e6c2c * tests: Added new test that try to connect/disconnect a same device
fcedfb9 * gattlib_adapter: Improve BLE scan code with comments and renaming
b57d954 * gattlib_glib_mainloop: Ensure mainloop is initialized before using it

oliviermartin, 2024-04-10:
You are missing the last commits: https://github.com/labapart/gattlib/commits/master/. You should do a git fetch or pull.

e665107, 2024-04-10:
Head:     master Add support to retrieve all manufacturer data from GATT advertisement packets
Merge:    origin/master Add support to retrieve all manufacturer data from GATT advertisement packets
Push:     origin/master Add support to retrieve all manufacturer data from GATT advertisement packets
Tag:      0.6.0 (11)

Stashes (1)
stash@{0} On master: temp0

oliviermartin, 2024-04-10:
Now, you need to rebuild `gattlib` - and modify your code (check example https://github.com/labapart/gattlib/blob/master/examples/advertisement_data/advertisement_data.c)

oliviermartin, 2024-04-12:
@e665107 , I close your issue as your original problem has been solved.

closed 2024-04-12 by oliviermartin